ifm/ifm3d #319: Timeout waiting for camera when get the bag

GoodTekken, 2021-07-29:
Hello,
I want to record the .bag from camera and not limit by time in Virtual Machine.
How can I do with this requirement.

When I use the instruction of " roslaunch ifm3d camera.launch " ,screen shows "Timeout waiting for camera".
'''
[ INFO] [1627538327.000199408]: Running dtors...
[ INFO] [1627538327.001237696]: Initializing camera...
[ INFO] [1627538328.016233543]: Initializing framegrabber...
[ INFO] [1627538328.034695069]: Initializing image buffer...
[ WARN] [1627538328.537653889]: Timeout waiting for camera!
[ WARN] [1627538329.041006533]: Timeout waiting for camera!
[ WARN] [1627538329.542305473]: Timeout waiting for camera!
[ WARN] [1627538330.044126430]: Timeout waiting for camera!
[ WARN] [1627538330.545792033]: Timeout waiting for camera!
[ WARN] [1627538331.047644623]: Timeout waiting for camera!
[ WARN] [1627538331.548749627]: Timeout waiting for camera!
[ WARN] [1627538332.049177906]: Timeout waiting for camera!
[ WARN] [1627538332.551014183]: Timeout waiting for camera!
[ WARN] [1627538333.053561812]: Timeout waiting for camera!
[ WARN] [1627538333.053594511]: Attempting to restart framegrabber...
[ INFO] [1627538333.053603150]: Running dtors...
[ INFO] [1627538333.054298912]: Initializing camera...
[ INFO] [1627538334.069874392]: Initializing framegrabber...
[ INFO] [1627538334.084131263]: Initializing image buffer...
[ WARN] [1627538334.585106067]: Timeout waiting for camera!

'''

And ping "192.168.0.69",it is okay.
ifm@ubuntu:~$ ping 192.168.0.69
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.
64 bytes from 192.168.0.69: icmp_seq=1 ttl=128 time=1.40 ms
64 bytes from 192.168.0.69: icmp_seq=2 ttl=128 time=1.36 ms
64 bytes from 192.168.0.69: icmp_seq=3 ttl=128 time=1.55 ms
64 bytes from 192.168.0.69: icmp_seq=4 ttl=128 time=1.57 ms

ifm@ubuntu:~$ ifm3d dump
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "561321355",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "0",
          "ClippingBottom": "263",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "351",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "false",
          "ContinuousUserFrameCalibration": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "true",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "false",
          "EnableRectificationDistanceImage": "false",
          "ExposureTime": "10000",
          "ExposureTimeList": "1250;10000",
          "ExposureTimeRatio": "8",
          "FrameRate": "5",
          "MaxAllowedLEDFrameRate": "5.4",
          "MinimumAmplitude": "42",
[... 136 lines cut ...]
      "Date": "Thu Jul 29 02:03:22 2021",
      "HWInfo": {
        "Connector": "#!02_A300_C40_04284835_008023176",
        "Diagnose": "#!03_D322_C34_04280777_008026824",
        "Frontend": "#!02_F342_C34_01066777_008026210",
        "Illumination": "#!02_I300_001_04246337_008001175",
        "MACAddress": "00:02:01:41:47:B0",
        "Mainboard": "#!03_M381_C41_04215040_008023690",
        "MiraSerial": "8470-6080-0175-06d8"
      },
      "SWVersion": {
        "Algorithm_Version": "2.1.9",
        "Calibration_Device": "00:02:01:41:47:b0",
        "Calibration_Version": "1.0.2",
        "Diagnostic_Controller": "v1.0.12-1a0474a2ad-dirty",
        "IFM_Recovery": "O3D3XX 0.6.73",
        "IFM_Software": "1.25.4073",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00029-gec409acac564-dirty (jenkins@dettlx190) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Sep 12 10:13:53 CEST 2018",
        "Main_Application": "1.0.12",
        "OEM_Software": "2.6.0"
      },
      "ifm3d_version": 1700
    }
  }
}

lola-masson, 2021-07-29:
Hi @GoodTekken,
Just to make sure I get this right:
You are trying to connect to an O3D camera from INSIDE an ubuntu VM on a Windows host?
The ping works from inside the VM or from the host? And the ifm3d dump?
It seems that you want to use ROS 1 to record bags?

GoodTekken, 2021-07-30:
Yes，I am trying to connect to an O3D camera from INSIDE an Ubuntu VM on a Windows host.
The ping works from inside the VM and the  ifm3d dump is also inside the VM.
I have a VM from IFM too. but just can record 10 seconds. 
So if I want 2 minute's record bags，How can I do for it?

graugans, 2021-07-30:
Hey @GoodTekken , it looks like your camera has configured two applications and one is configured to [software trigger](https://github.com/ifm/ifm3d/blob/master/modules/camera/include/ifm3d/camera/camera.h#L100):

The first one
```
 ifm3d dump | jq '.ifm3d.Apps[0].TriggerMode'
"2"
```
The second one
```
 ifm3d dump | jq '.ifm3d.Apps[1].TriggerMode'
"1"
```
Please note we are counting from 1 upwards on the camera
```
 ifm3d dump | jq '.ifm3d.Device.ActiveApplication'
"1"
```
To verify this assumption your could try this 
```
ifm3d hz --nframes=100 -sw
```

GoodTekken, 2021-07-30:
Yes,Thank you,.
it shows "Timeout waiting for camera!".
So how to do the next to get  the record bag ?

Looking forware to your reply.

ifm@ubuntu:~$ ifm3d hz --nframes=100 -sw
Timeout waiting for camera!
ifm@ubuntu:~$ ifm3d hz --nframes=100 -sw
Timeout waiting for camera!

graugans, 2021-07-30:
Hey @GoodTekken ,

please try:
```
ifm3d hz --nframes=100 --sw
```
I had a typo in the first response

GoodTekken, 2021-08-02:
Thank you @graugans,
It helps a lot and succeeds for recording.

The result below:
tekken@tekken-virtual-machine:~$ ifm3d hz --nframes=100 --sw
FrameGrabber running at: 2.76904 Hz
100 frames captured, over 1 runs

closed 2021-08-02 by GoodTekken

Gabiriyal, 2023-06-17:
hi ,i have an this same issue in configure the O3R camera 

[ INFO] [1686975141.666864385]: Initializing camera...
[ INFO] [1686975146.677230126]: Initializing framegrabber...
[ INFO] [1686975151.677437140]: Start streaming frames
[ WARN] [1686975156.678626410]: Timeout waiting for camera!
[ WARN] [1686975156.678666220]: Attempting to restart framegrabber...
[ INFO] [1686975156.678681432]: Running dtors...
[ INFO] [1686975156.678854796]: Initializing camera...
[ INFO] [1686975161.691967945]: Initializing framegrabber...
[ INFO] [1686975166.692378127]: Start streaming frames
[ WARN] [1686975171.693039815]: Timeout waiting for camera!
[ WARN] [1686975171.693129855]: Attempting to restart framegrabber...
[ INFO] [1686975171.693171770]: Running dtors...
[ INFO] [1686975171.693577726]: Initializing camera...
[ INFO] [1686975176.705947934]: Initializing framegrabber...
[ INFO] [1686975181.706168308]: Start streaming frames
[ WARN] [1686975186.707217287]: Timeout waiting for camera!
[ WARN] [1686975186.707316279]: Attempting to restart framegrabber...
[ INFO] [1686975186.707379356]: Running dtors...
[ INFO] [1686975186.707818487]: Initializing camera...
[ INFO] [1686975191.722609885]: Initializing framegrabber.

GoodTekken, 2023-06-25:
@Gabiriyal Hello， friend， would you Ping the camera IP in your computer?
1.  Would you use a stable supply voltage for the camera?
2.  Would you try the instruction of ` ifm3d dump`  to get the configuration?
3. Would you try  the instruction of  `ifm3d hz --nframes=100 --sw` to get the camera feedback?

ismailalhajosman, 2024-06-21:
Hello @GoodTekken ! 
Sorry to pup up to this issue but I have a camera with the same version as yours (`03DP03`) . Although I tried many combinations of` ifm3d` and `ifm3d - ros` driver to run it but it always fails to run the` camera.launch` . I would appreciate if you could help me run the camera by following the steps you have done to make it work !

dekhanra, 2024-06-21:
Hello @ismailalhajosman

The O3DPxx camera is Pallet Detection Application camera and works triggers the camera for every command. Could you let me know how is your software setup looks like i.e.

- Firmware on O3DP03
- ROS1 version
- ifm3d-ros version

>Note: The latest ifm3d-ros versions are compatible only with the O3R cameras

ismailalhajosman, 2024-06-21:
Hi @dekhanra !
My setup looks like this: 

- Firmware on O3DP03 : `2.10.15`.
- ROS1 version : `Noetic `.
- ifm3d-ros version : tried `1.1.2`, `1.1.1` and `0.7.0` versions but none of them worked.
- I also tried ifm3d version of `1.5.3` , `1.3.3`,  `1.2.6 `,  and `0.20.0` in combination with `ifm3d - ros` but none of them worked.