ifm/ifm3d #320: cmake could not find xmlrpc-c config.cmake file

scoperesolutor, 2021-08-21:
Hello, I am trying to build the files for the ifm3d but I seem to be getting stuck on one of the modules which gives the error below. 

"CMake Error at modules/camera/src/libifm3d_camera/CMakeLists.txt:26 (find_package):
Could not find a package configuration file provided by "xmlrpc-c" with any
of the following names:

xmlrpc-cConfig.cmake
xmlrpc-c-config.cmake

Add the installation prefix of "xmlrpc-c" to CMAKE_PREFIX_PATH or set
"xmlrpc-c_DIR" to a directory containing one of the above files. If
"xmlrpc-c" provides a separate development package or SDK, be sure it has
been installed."

inbangsa, 2021-08-22:
can you please specify OS ?

scoperesolutor, 2021-08-22:
oops.  windows 10 x64. I did follow the instructions to build the dependencies individually. Version	10.0.18363 Build 18363

inbangsa, 2021-08-23:
@scoperesolutor

I assume that you have installed ifm3d and dependencies at ```C:/ifm3d/install```  as per the [build instruction](https://github.com/ifm/ifm3d/blob/master/doc/windows.md)
 
can you check ```C:/ifm3d/install/lib/cmake/xmlrpc-c``` this should have xmlrpc-c-config.cmake . if you donot have this file then please follow the [build instruction for xmlrpc-c](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#xmlrpc-c) and install the libs. 

if you have these please share your  ```cmake``` command used to configure the ifm3d build.  we need to provide install path of libs as a value to CMAKE_PREFIX_PATH, so that cmake can find this libraries.

scoperesolutor, 2021-08-23:
@inbangsa 

There is a config with a header that is close to the target name; "xmlrpc-config.h.cmake" but otherwise no i cannot find that in xmlrpc-c folder.
[image]

inbangsa, 2021-08-23:
@scoperesolutor 

Please build the library, this are source file from the clone, Please follow all [instruction for source build](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#source-dependencies)

scoperesolutor, 2021-08-23:
Thanks. I found the file as you outlined above. I directed cmake to that folder and that particular error is gone now. 
I now have the failed to run command:

    MSBuild.exe

  to get the value of VCTargetsPath:      Probably because updated to the latest cmake. i will downgrade and see what happens.

closed 2021-08-23 by scoperesolutor