ifm/ifm3d #321: Is the camera's light center the origin of the camera's coordinate system, for example, should the X and Y values directly in front of the camera's light center be zero? I placed a plane region directly in front of the camera plane, but the X and Y values of the position of the point directly opposite the camera light center are not 0. Why is this?

haoyuefan, 2021-09-10:


haoyuefan, 2021-09-13:
@graugans

graugans, 2021-09-14:
Hey @haoyuefan , I was out of office the last three weeks.

What kind of plane did you put in front of the camera? What we typically define as the center of the coordinate system is the front glass of the camera and the center of the front face. But @GreNait or @lola-masson may have some graphics to explain this in more detail.

Johnlan197, 2021-11-24:
@graugans what's the defination of coordinate system of camera? ,specifically,where are the direction of X,Y,Z axis? i cannot find any technical reports in @GreNait or @lola-masson.

ifm-csr, 2021-11-24:
Hi @Johnlan197 ,

In principle, our systems use Z for the distance, X for the left/right and Y for up/down. However, the ifm3d is using (I think) the robotic approach. meaning that X is the distance. Does that make sense? 

All coordinates are in reference to the camera center. Which is depending on the camera. For the O3D it is at the middle of the lens at the height of the orange sealing. You can of course provide an extrinsic calibration if you want to. 

Does that make sense?

GreNait, 2021-11-24:
Hi @Johnlan197 ,

In principle, our systems use Z for the distance, X for the left/right and Y for up/down. However, the ifm3d is using (I think) the robotic approach. meaning that X is the distance.

All coordinates are in reference to the camera center. Which is depending on the camera. For the O3D it is at the middle of the lens at the height of the orange sealing. You can of course provide an extrinsic calibration if you want to.

Does that make sense?

flyaudio, 2022-10-11:
> 

is there a more intuitional picture showing the point cloud origin relative to camera shell