ifm/ifm3d #325: Failed to build and install xmlrpc-c

mankiet, 2021-10-05:
Hi All,

I'm trying to install ifm3d in my windows OS and I have encountered the following error during the build and install of xmlrpc-c.

Error:

> xmlrpc_client.lib(xmlrpc_curl_transport.obj) : error LNK2019: unresolved external symbol __imp_WSAStartup referenced
 in function setupGlobalConstants [C:\ifm3d\xmlrpc-c\build\tools\xml-rpc-api2cpp\xml-rpc-api2cpp.vcxproj]
xmlrpc_client.lib(xmlrpc_curl_transport.obj) : error LNK2019: unresolved external symbol __imp_WSACleanup referenced
 in function setupGlobalConstants [C:\ifm3d\xmlrpc-c\build\tools\xml-rpc-api2cpp\xml-rpc-api2cpp.vcxproj]
xmlrpc_util.lib(select.obj) : error LNK2019: unresolved external symbol __imp_select referenced in function xmlrpc_p
select [C:\ifm3d\xmlrpc-c\build\tools\xml-rpc-api2cpp\xml-rpc-api2cpp.vcxproj]
C:\ifm3d\xmlrpc-c\build\tools\xml-rpc-api2cpp\Release\xml-rpc-api2cpp.exe : fatal error LNK1120: 3 unresolved extern
als [C:\ifm3d\xmlrpc-c\build\tools\xml-rpc-api2cpp\xml-rpc-api2cpp.vcxproj] 
>

I'm using windows 10 pro and my OS build is 19042.1237. Do you have any idea about this?

mankiet, 2021-10-06:
Hi @inbangsa ,

I'm using Visual Studio 2019 to build the xmlrpc-c.

Following is my variables:
IFM3D_BOOST_ROOT="C:\Program Files\PCL 1.8.1\3rdParty\Boost"
IFM3D_OPENCV_PATH=C:\opencv\build
IFM3D_CMAKE_GENERATOR="Visual Studio 16 2019"
IFM3D_BUILD_DIR=C:\ifm3d 
CONFIG=Release

I'm encounter the error after this command `cmake -G %IFM3D_CMAKE_GENERATOR% -Ax64 -DCMAKE_INSTALL_PREFIX=%IFM3D_BUILD_DIR%\install ..`

inbangsa, 2021-10-06:
@mankiet Thanks for the issue, I am able to reproduce this, recently we updates the xmlrpc-c with suppport for static builds and we are in process to update the ifm3d documentation for xmlrpc-c build.  

to resolve your issue please add ```-DBUILD_SHARED_LIBS=ON``` in your cmake configure command. 

complete command is as follow: 
```cmake -G %IFM3D_CMAKE_GENERATOR% -Ax64 -DCMAKE_INSTALL_PREFIX=%IFM3D_BUILD_DIR%\install -DBUILD_SHARED_LIBS=ON .. ```

Please let me know if this works for you.

mankiet, 2021-10-12:
Hi @inbangsa,

Yes. Your solution is working for me. However, when I proceed to install and build ifm3d I have encountered the followings error.

> -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
CMake Error at CMakeLists.txt:75 (add_subdirectory):
  The source directory
C:/ifm3d
does not contain a CMakeLists.txt file.
CMake Error at C:/ifm3d/xmlrpc-c/cmake/xmlrpc-c-config.cmake:1 (include):
  include could not find requested file:
C:/ifm3d/xmlrpc-c/cmake/xmlrpc-c-targets.cmake
Call Stack (most recent call first):
  modules/camera/src/libifm3d_camera/CMakeLists.txt:26 (find_package)
-- OpenCV ARCH: x64
-- OpenCV RUNTIME: vc15
-- OpenCV STATIC: OFF
-- Found OpenCV 3.4.9 in C:/opencv/build/x64/vc15/lib
-- You might need to add C:\opencv\build\x64\vc15\bin to your PATH to be able to run your applications.
-- looking for PCL_COMMON
-- OpenCV ARCH: x64
-- OpenCV RUNTIME: vc15
-- OpenCV STATIC: OFF
-- Found OpenCV 3.4.9 in C:/opencv/build/x64/vc15/lib
-- You might need to add C:\opencv\build\x64\vc15\bin to your PATH to be able to run your applications.
-- Cannot find dpkg in your path, default to AMD64.
-- Configuring incomplete, errors occurred!
See also "C:/ifm3d/ifm3d/build/CMakeFiles/CMakeOutput.log".
See also "C:/ifm3d/ifm3d/build/CMakeFiles/CMakeError.log".

Any idea what caused this error? I also added C:\opencv\build\x64\vc15\bin in my env path.

mankiet, 2021-10-12:
Hi @inbangsa,

Yes. I have successfully installed xmlrpc and my IFM3D_BUILD_DIR is "C:\ifm3d".
However, I don't see there is an install folder in my IF3D_BUILD_DIR and I found xmlrpc-c-config.cmake in C:\ifm3d\lib\cmake\xmlrpc-c.

I also have tried change the -DCMAKE_PREFIX_PATH to C:\ifm3d and C:\ifm3d\lib. Unfortunately, the error is still there.

inbangsa, 2021-10-12:
@mankiet the issue is you installed and source is at c:\ifm3d and now cmake is defauling to source xmlrpc-config.cmake then lib\cmake\xmlrpc-c.

you need to fix this by re-installing xmlrpc to install folder.

and then passing %IFM3D_BUILD_DIR%\install to -DCMAKE_PREFIX_PATH.