ifm/ifm3d #330: fg.wait_for_frame(im,1000) always return false

PHILCHEN666, 2021-10-31:
now,i have the ifm3dpy,it can run in the linux 20.04 focal,In the VMware,i also can get data from (ping 192.168.0.69),it show we already connect to the camera,but always return false in the code "fg.wait_for_frame(im,1000)".(fg = ifm3dpy.FrameGrabber(cam)),how could i slove this problem,thanks for your patient guidance.

lola-masson, 2021-11-01:
You might need to update your firmware. Which version is currently on your O3D303?

PHILCHEN666, 2021-11-01:
ifm_VisionAssistant show the Firmware ver 1.6.2114

lola-masson, 2021-11-01:
You can download the new FW (we are at version 1.71.x) on [ifm.com](https://www.ifm.com/us/en/product/O3D303?tab=documents). You will need to create a (free) account. Please update your O3D and check if you have the same problem as before.

PHILCHEN666, 2021-11-02:
thank you ,it can work now

PHILCHEN666, 2021-11-03:
hello, i want know the xyz_image capture by the ifm3dpy. the value in that image not the float (for example -1.351,2.31 etc),but is (-652,256),what the relationship about the value in real x distance and the image value.same question in amp_image,rdis_image,yz_image.

graugans, 2021-11-04:
Hey @PHILCHEN666 did I get you right, the issue is solved? In this case please close this issue as-well.

closed 2021-11-05 by PHILCHEN666