ifm/ifm3d #331: Cannot connect to O3R Camera using ifm3d library

mehranq, 2021-11-09:
Hello. I am trying to test out the O3R camera using the C++ example provided here: https://ifm.github.io/ifm3d-docs/examples/o3r/configuration/configuration.html. I am using the Windows platform and built the ifm3d library from source using your "Building ifm3d from source on Windows" guide without the PCL dependency and using the o3r/main-next branch. 

The camera is connected to my pc and I am able to ping it successfully at **192.168.0.69** through command prompt on windows.

However when I try to connect to the camera using the example program in the link above it still fails to connect to the camera and I get the an exception when it gets to the line `json conf = cam->Get();`. See attached screenshot titled as **"error without explicitly specifying ip address of camera"**.  

Interestingly, if i explicitly define the ip address and port no using `auto cam = std::make_shared<ifm3d::O3RCamera>("192.168.0.69", ifm3d::DEFAULT_XMLRPC_PORT);` in the cam object constructor it throws a different exception right there at the camera object declaration line and the exception source is in a file called **xutility**. It does not seem to matter whether the above explicit ip address is a valid one or not. I tried passing an invalid ip address like **192.168.0.50** and it would still throw the same exception in **xutility**. See attached screenshot titled as **"error when explicitly specifying ip address of camera"**. 

Hope you can advise on what the issue might be. The camera does respond successfully when pinging it through the command prompt but it just does not seem to connect through the example C++ program.
[image]
[image]

graugans, 2021-11-09:
Hey @mehranq ,
what Visual Studio version are you using?

inbangsa, 2021-11-09:
@mehranq  are you building example in Debug mode ? 
can you try building example in Release mode,

by default ifm3d is build in release mode.

mehranq, 2021-11-09:
> Hey @mehranq , what Visual Studio version are you using?

I am using Visual Studio 2019.

mehranq, 2021-11-09:
> @mehranq are you building example in Debug mode ? can you try building example in Release mode,
> 
> by default ifm3d is build in release mode.

Yes. I am building it debug mode. I'll try building the example in release mode. Thank you for the suggestion.

mehranq, 2021-11-09:
> @mehranq are you building example in Debug mode ? can you try building example in Release mode,
> 
> by default ifm3d is build in release mode.

I just tried building in release mode. I am still getting an exception on line  `json conf = cam->Get();`. However the exception type changed to `ifm3d::error_t` whereas previously when running in debug mode, the exception type was `std::bad_array_new_length`

Also, I don't get the exception in **xutility** anymore when explicitly specifying the ip address and port  for the camera.  Instead the code does continue running and throws the same above `ifm3d::error_t` exception on the line `json conf = cam->Get();`

So now, I get the same exception regardless of whether I explicitly specify the ip address and port now in camera constructor or not.  Here is a screenshot of the exception:

[image]

inbangsa, 2021-11-09:
@mehranq can you please provide dump of the device configuartion.

you need to do this with ifm3d tool which is installed in your PC while installing ifm3d. you will find this tool in ```<install prefix path>/bin```

getting the dump :

> ifm3d dump 

this will help us to understand the configuration of your device and firmware that you are using.

mehranq, 2021-11-09:
I tried using ifm3d dump as instructed but that command throws an error like this

[image]
:

When adding the try catch snippet above to the C++ example code, the error code seems to be the same as the one I get when trying to run ifm3d dump:

[image]

desengph, 2021-11-10:
Hi @mehranq 
this sounds "suspiciously" similar to a mismatch of firmware and software versions to me. 

Can you please check that you are using a recent O3R VPU firmware version (version >= 0.13.11). The easiest way to check this is the use of our web front-end: `http://192.168.0.69/api/rpc/v1/com.ifm.efector/`. Copy and paste the `getSWVersion` command into the second box from the top, label method. Any other box stays doesn't need any user input for this command.

Also can you please double check that your power supply is capable of supplying the required 5 Amp at 24V.

graugans, 2021-11-10:
Hi, @mehranq,

like @desengph mentioned you may have some old O3R firmware running. The easiest way to query the firmware version is pasting this line in the address bar of your browser:

```
http://192.168.0.69/api/rpc/v1/com.ifm.efector/?method=getSWVersion
```

The result looks like this:
```
Response
{ "error": null, "id": null, "result": { "IFM_Software": "0.14.0-242", "Linux": "Linux version 4.9.140-l4t-r32.4+gc35f5eb9d1d9 (oe-user@oe-host) ", "Main_Application": "0.16.0" } }
```
In your case there maybe an older FW version

mehranq, 2021-11-10:
Hi @graugans @desengph
Here is the result of the api call that you suggested to check the firmware version:

[image]

It does look like the firmware version is old on my end. How should I go about updating the firmware?

Also, just to confirm, I am using the following power supply supplied by IFM. The output is rated at 3A 24V. Just FYI, we only have one camera connected to the VPU so I would assume it should be enough power:

[image]

desengph, 2021-11-10:
Hi @mehranq 
I'm sorry to say but your firmware on the O3R camera is ancient. 
We will have to see if we can update your hardware without risking a software hardware incompatibility for early field test hardware.

@mehranq can you please send me a PM. We will have to discuss the details in private.

I'm sorry for the confusion earlier.

Best regard,
Philipp

mehranq, 2021-11-10:
@desengph 
Sure. How do I get in touch with you privately?

desengph, 2021-11-10:
@mehranq 
I just relalized that GitHub removed their direct messages service.
Can please send me an email philipp.senger<at>ifm.com

Otherwise let me know your email address so I can shoot you an email.

mehranq, 2021-11-10:
@desengph 
I just shot you a quick email. Thanks.

desengph, 2021-11-10:
Got it. 
I will collect the necessary information from my Canadian colleagues and answer you tomorrow.

mehranq, 2021-11-10:
Great. Thank you.

inbangsa, 2021-11-18:
@mehranq if your issue is resolved can you please close this.

mehranq, 2021-11-18:
@inbangsa sure. Closing it now.

closed 2021-11-18 by mehranq