ifm/ifm3d #332: ifm3d ox3d100使用过程接口，程序会崩溃

zchao9456, 2021-11-17:
#include <iostream>
#include <memory>
#include <opencv2/core/core.hpp>
#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>
#include <ifm3d/pcicclient.h>
const std::string ifm3d::DEFAULT_PASSWORD = "";
const std::uint16_t ifm3d::DEFAULT_XMLRPC_PORT = 80;
const int ifm3d::DEFAULT_PCIC_PORT = 50010;
const std::uint16_t ifm3d::DEFAULT_NAT_PCIC_PORT = 0;
const std::string ifm3d::DEFAULT_IP = std::getenv("IFM3D_IP") == nullptr ? "192.168.0.69" : std::string(std::getenv("IFM3D_IP"));
const int ifm3d::MAX_HEARTBEAT = 300; // secs
const std::size_t ifm3d::SESSION_ID_SZ = 32;
const std::string ifm3d::DEFAULT_APPLICATION_TYPE = "Camera";

int main(int argc, const char **argv)
{
	auto cam = ifm3d::Camera::MakeShared(ifm3d::DEFAULT_IP, ifm3d::DEFAULT_XMLRPC_PORT, ifm3d::DEFAULT_PASSWORD);
	auto fg = std::make_shared<ifm3d::FrameGrabber>(cam);
	auto im = std::make_shared<ifm3d::ImageBuffer>();
	// Create pcic interface and initialize configuration

	ifm3d::PCICClient::Ptr pcic = std::make_shared<ifm3d::PCICClient>(cam);
	////// Set notification (and error) callbacks, which simply print received messages

	pcic->SetNotificationCallback([](const std::string& notification)
	{
		std::cout << "Notification: " << notification << std::endl;
	});

	pcic->SetErrorCallback([](const std::string& error)
	{
		std::cout << "Error: " << error << std::endl;
	});

	//// Switch between applications (and receive notification 000500000)
	//std::cout << "Switch to application 1" << std::endl;

	pcic->Call("a01");

	cv::Mat amp;
	cv::Mat xyz;

	while (true)
	{
		//pcic->Call("t");
		if (!fg->WaitForFrame(im.get(), 1000))
		{
			std::cerr << "Timeout waiting for camera!" << std::endl;
			//return -1;
		}

		amp = im->AmplitudeImage();
		xyz = im->XYZImage();
		pcl::PointCloud<pcl::PointXYZI>::Ptr _clouds = im->Cloud();
		printf("Pts Size(%d)\n", _clouds->points.size());
		// now do something with `amp` and `xyz`
	}

	return 0;
}

ox3d100能不能使用过程接口，通过软件触发的模式来获取数据，当我把ox3d100配置成连续模式时，用上面的代码能正常获取到数据，当改成过程接口时，程序会在这里发生中断ifm3d::PCICClient::Ptr pcic = std::make_shared<ifm3d::PCICClient>(cam);，请问这样的问题怎么解决呢。同样用上面的代码，使用ox3d300相机，一切都正常，这是什么原因呢，是不是两种相机，触发模式不一样？麻烦帮助一下，谢谢

graugans, 2021-11-17:
Hey @zchao9456 ,

my Chinese is kind of none existing, I ran your question through Google translate:

>Can I use the process interface to obtain data through the mode triggered by the software? When I configure the ox3d100 to continuous mode, the data can be obtained normally with the above code. When the process interface is changed, the program will be interrupted here ifm3d: :PCICClient::Ptr pcic = std::make_sharedifm3d::PCICClient(cam);, how can I solve this problem? Using the same code above and using the ox3d300 camera, everything is normal. What is the reason? Is the trigger mode different for the two cameras? Please help, thank you

Are you using an O3X100 and an O3D300? The O3X100 does not provide support for the `PCICClient` on O3X1xx we only do have a limitid PCIC implementation. There is only sending of data implemented, it also only does support a single client. When connecting the PCICClient the single channel will be occupied by this instance.

What information are you looking for on the `PCICClient`? Maybe this is available through a different channel.

zchao9456, 2021-11-17:
yes, O3X100 and O3d300,Can the o3x100 use the process interface? I don't want to take pictures continuously. I want to take pictures when I need it. Can the o3x100 support this? Thank you @graugans

graugans, 2021-11-17:
Hey @zchao9456 , yes the O3X1xx provides a software trigger. With this software trigger you can control when images are acquired.

graugans, 2021-11-17:
Here is a link to the [implementation](https://github.com/ifm/ifm3d/blob/master/modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp#L211)

graugans, 2021-11-17:
And an [example](https://github.com/ifm/ifm3d-examples/blob/master/multi_camera_grabber/ex-multi_camera_grabber.cpp#L99) how to use it

zchao9456, 2021-11-17:
Thank you @graugans

closed 2022-10-23 by github-actions[bot]