ifm/ifm3d #333: Extracting X, Y, Z point cloud values from im->XYZImage()

mehranq, 2021-11-17:
I am writing the following function  to extract the X,Y,Z point cloud values from the image returned by im->XYZImage() so I can call this function in my own program and parse the point cloud values accordingly. 

I would like to get your feedback on whether this is the correct approach because I am a bit unsure on the data structure returned by XYZImage() function.  Please feel free to make any suggestions or improvement. Thank you.

`bool Get3DPointCloud(ifm3d::O3RCamera *cam, float* x, float* y, float* z, uint32_t timeout, int pcic_port) {
    if (cam != nullptr) {
   
        try {
            // Block of code to try
            auto fg = std::make_shared<ifm3d::FrameGrabber>((ifm3d::CameraBase::Ptr)cam, ifm3d::DEFAULT_SCHEMA_MASK, pcic_port);
            auto im = std::make_shared<ifm3d::StlImageBuffer>();
            //////////////////////////
            // Get a frame:
            //////////////////////////
            if (!fg->WaitForFrame(im.get(), timeout))
            {
                std::cerr << "Timeout waiting for camera!" << std::endl;
                return false;
            }
            //////////////////////////
            // Get 3D Image:
            //////////////////////////
            auto pclImage = im->XYZImage();

            int k = 0;

            for (unsigned int i = 0; i < pclImage.height(); i++)
             {
                 for (unsigned int j = 0; j < pclImage.width(); j++)
                 {
                     const uint8_t* rowcolij = pclImage.ptr<uint8_t>(i,j);
                     x[k] = rowcolij[0];
                     y[k] = rowcolij[1];
                     z[k] = rowcolij[2];
                     k++;
                 }
             }
            std::cout << pclImage.height() << " " << pclImage.width() << std::endl;
        }
        catch (const ifm3d::error_t& ex) {
            // Block of code to handle errors
            std::cout << "Failed to perform gaging operation. \n";
            std::cerr << "ifm3d error: " << ex.code() << std::endl << ex.what() << std::endl;
            return false;
        }     
        return true;
    }
    else {
        std::cout << "Failed to get camera object. \n";
        return false;
    }`

inbangsa, 2021-11-18:
@mehranq  ```ifm3d::Image``` stores data in contiguously, and it provides ```nchannels()``` and ```dataformat()``` to know which type of data is stored in the ```ifm3d::Image```. you can find more information of dataformat [here](https://github.com/ifm/ifm3d/blob/o3r/main-next/modules/framegrabber/include/ifm3d/fg/byte_buffer.h#L35).

for xyz Image the nchannels is 3 and data type is 6 (ifm3d::data_format::FORMAT_32F) which defines each pixel/point has of 3 values and of type float. 

following code shows methods to extract data
```Cpp

auto image = img->XYZImage();

std::cout << image.nchannels() << std::endl;
std::cout << static_cast<int>(image.dataFormat()) << std::endl;

 // with point3D_32F 
 ifm3d::Image_< ifm3d::Point3D_32F>  xyz = image; // no copy of data 

// range based for loop
  for (const auto& point : xyz)
  {
    std::cout << "(" << point.val[1] << "," << point.val[2] << "," << point.val[0] << ")";
  }

  // with pointer 
  for (int i = 0; i < image.height(); i++)
  {
    for (int j = 0; j < image.width(); j++)
    {
      auto ptr = image.ptr<float>(i, j);
      auto z = ptr[0];
      auto x = ptr[1];
      auto y = ptr[2];

      std::cout << "(" << x << "," << y << "," << z << ")";

    }
  }

```
Other optimization that I can suggest is to not create ```ifm3d::FrameGrabber``` and ```ifm3d::StlImageBuffer``` outside your Get3DPointCloud function. I assume that you are going to call this function under a tight loop.  creating FrameGrabber for every frame capture will need time for connection to device and we recommended to keep creation of this objects out of tight loop.

mehranq, 2021-11-18:
@inbangsa Thank you for the answer. It seems what I was doing in my code was mostly correct.  I will keep the optimizations you mentioned in mind. Just a couple of follow up questions based on your code above:

1) Each point/pixel contains the point cloud values with Z value as the first index followed by X and Y, right? Instead of it being in X, Y and Z order like in my code? Just making sure it is not a typo in your code :)

2) The image dimensions returned by image.height() and image.width() above would be determined by the resolution height and width values of the camera head that is returning the frame, right?

inbangsa, 2021-11-19:
@mehranq
1) There are some legacy frame conversion due to which  Z,X,Y at index 0,1,2 respectively, There is already a work in progress which will provide compile time switch to disable these conversion and XYZ will be available at 0,1,2 index. I inform here when we release this change .
2) yes they determine the resolution of height and width values of camera head for 3D ports. for 2D port we provide JPEG encoded data as a data blob. actually resolution of 2D images can be obtain after decoding the IPEG data.

closed 2021-12-07 by mehranq