ifm/ifm3d #334: Error when trying to use device discovery (O3R camera)

mehranq, 2021-12-03:
Hi. A couple of things I want to raise/ask.

**1)** I am trying to using the DeviceDiscovery function from the camera base class to find the O3R cameras connected to my VPU. I am getting the following exception when trying to do so. When I check the console it has the line, "An invalid argument was supplied" printed. I have tried running is directly using the ifm3d command line tool using "ifm3d discover" command but I get the same "An invalid argument was supplied"  error message.
[image]

I am able to run ifm3d dump and here is the result for that:
[ifm3d dump.txt](https://github.com/ifm/ifm3d/files/7652226/ifm3d.dump.txt)

**2)** I would like to confirm whether the following (highlighted in red squares) are LEDs that are supposed to come one when a camera is connected in the corresponding port. In my case, they don't come on when I  have a camera connected. 
[image]

Hope I can get some help on these issues.

graugans, 2021-12-06:
Hey @mehranq , the LEDs are supported in the next firmware release series 0.14.x which we currently do internal testing with.

inbangsa, 2021-12-06:
@mehranq on discovery:  I have tried to reproduce the issue but was not able to reproduce it on windows nor on Linux. can you provide more information on which windows and Visual studio version you are using and would also want to know the build type of ifm3d (Release/Debug)

ifm3d discover uses udp communication for discovery of devices in subnet. Are udp packets allowed on your PC and not blocked by firewall on network. can you try creating a outbound and inbound rule on firewall for udp port 3321 ( this port is used by ifm3d for discovery).

mehranq, 2021-12-06:
> Hey @mehranq , the LEDs are supported in the next firmware release series 0.14.x which we currently do internal testing with.

@graugans Thank you for the info. 

@inbangsa Here is the info you requested:

**Windows version:**
[image]

**Visual Stuido version:**
[image]

I checked the firewall settings and the inbound rules were already setup like so:
[image]

I added the outbound rule as per your suggestion like so:
[image]

I restarted my pc and re-ran the ifm3d discover command. However, I still get the same error message  "An invalid argument was supplied"

FYI, I built the library from source on windows in **RELEASE** config as instructed on your github page for building on windows. I built the library about a month ago. Could it be the case that this issue existed previously but does not occur in the latest version of the library?

inbangsa, 2021-12-21:
@mehranq there is no change in this feature from last couple of months.

we are not able to reproduce this issue, and all tests are working fine on windows and linux machine. we are in developement of windows installer for ifm3d which will provide binary version and we would like you to test this feature with that.

closed 2022-10-23 by github-actions[bot]