ifm/ifm3d #335: How to determine whether an actual camera (O3R) is connected to the VPU

mehranq, 2021-12-07:
Hi. I am trying to use the following code to determine whether a camera is connected to the VPU or not. I am using an O3R camera:
`/*
 * Copyright 2021-present ifm electronic, gmbh
 * SPDX-License-Identifier: Apache-2.0
 */

#include <iostream>
#include <iomanip>
#include <memory>
#include <fstream>
#include <ifm3d/camera/camera_o3r.h>
#include <ifm3d/camera/err.h>
using json = nlohmann::json;

int main() {

    try {

        std::string ip_address = "192.168.0.69";
        uint16_t xmlrpc_port = 80;
        std::cout << "IP Address: " << " " << ip_address << std::endl;
        std::cout << "XMLRPC Port: " << " " << xmlrpc_port << std::endl;

        //probe hardware to determine if the connected cam is an O3R
        auto cam_unknown = ifm3d::CameraBase::MakeShared(ip_address, xmlrpc_port);

        std::string dev_fam = cam_unknown->DeviceType(false);
        std::cout << "Dev Fam: " << " " << dev_fam << std::endl;
        if (dev_fam == "1:768") {
            std::cout << "Found O3R. \n";
        }
        else {
            std::cout << "Unsupported camera type connected. \n";
        }
    }
    catch (const ifm3d::error_t& ex) {
        std::cout << "Failed to initialize camera. Make sure the camera hardware is connected. \n";
        std::cerr << "ifm3d error: " << ex.code() << std::endl << ex.what() << std::endl;
    }

    return 0;
}`

Using the above code I first turned on the VPU **without** having my O3R camera's 2D or 3D cable connected to the ports on the VPU. I ran the above code and the `cam_unknown->DeviceType(false);` returns the value "1:768" as a string store in `dev_fam`. I then turned off the VPU and connected my O3R camera to Port1(2D) and Port3(3D) and then turned the VPU back on and ran the code again. It still returns the same "1:768" value in `dev_fam`. 

Based on my understanding from your api documentation, the `DeviceType(false)` is supposed to probe the connected camera to determine the device type connected. But how come it returns the same value regardless of whether I have a camera connected to the VPU or not?

How can one determine whether an actual camera is connected to the VPU and further determine the type of camera connected (O3R, O3D, etc.)?

lola-masson, 2021-12-07:
Hi @mehranq, the DeviceType function is more for internal uses. We recommend using either the `WhoAmI` or the `AmI` functions like so:
```cpp
// Use this:
std::cout<< cam->AmI(ifm3d::CameraBase::device_family::O3R) << std::endl;
// Or this:
printf("%d", cam->WhoAmI() == ifm3d::CameraBase::device_family::O3R);
```

mehranq, 2021-12-07:
Hi @lola-masson
I did try using `WhoAmI` but had the same issue where it always returns device family as O3R regardless of whether I have any camera heads connected to the VPU or not. 

"Note that we refer to the O3R as the whole platform (VPU + _heads)."_
Based on the above line it makes sense then that `WhoAmI` always returns O3R regardless of whether I have any camera heads connected to the VPU or not. 

_"If you want to only find out about the connected heads, you will need to look through the json output of the dump (the type 2D/3D of each imager is contained in the variable "type")."_
I will try to determine whether a camera is connected based on the json dump then. 

Is it safe to assume that `cam->Get()` will only list the ports where a camera head is connected?

graugans, 2021-12-08:
Hey @mehranq,

> I did try using WhoAmI but had the same issue where it always returns device family as O3R regardless of whether I have any camera heads connected to the VPU or not.

This is intended, as the O3R consists of the OVP8xx and the connected heads, the Ethernet connection belongs to the OVP8xx which is also called the VPU this is what has the device type `1:768`

> Is it safe to assume that cam->Get() will only list the ports where a camera head is connected?
It will only list connected ports. There notion of a port does not necessarily be a camera, it can also be the IMU or any future sensing device.  So one at least may also want to check for the specific type of a port at least weather it is a 2D head or a 3D head.

mehranq, 2021-12-08:
Hi @graugans 
> It will only list connected ports. The notion of a port does not necessarily be a camera, it can also be the IMU or any future sensing device. So one at least may also want to check for the specific type of a port at least weather it is a 2D head or a 3D head.

Thank you for the clarification. Based on this, I will check the list of connected ports returned by `cam->Get()` and then filter it further based on "type". Thank you.

closed 2021-12-13 by mehranq