ifm/ifm3d #337: ifm3d\xmlrpc-c\lib\curl_transport\curltransaction.c(19,10): fatal error C1083: Unable to open include file: “curl/curl.h”: No such file or directory

zwenjiang, 2021-12-08:
Hi, 
   When I follow the steps by Doc /windows.md#running-ifm3d-tool-on-windows , I got an error compiling xmlrpc-c using VS2019, Prompt not found ifm3d\xmlrpc-c\lib\curl_transport\curltransaction.c(19,10): fatal error C1083: Unable to open include file: “curl/curl.h”: No such file or directory. What should I do about it? tks

inbangsa, 2021-12-09:
@zwenjiang it seems curl installation is missing on the system. from the information provided I understand that you are trying to build ifm3d from source with these [instruction](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#source-dependencies). 
To build and install curl please follow this [instruction](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#curl). and after this you should be able to build xmlrpc-c with these [instructions](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#xmlrpc-c) 

you might need to additionally pass -DBUILD_SHARED_LIBS=ON to cmake configure command of xmlrpc-c.  We will update this in documentation in our next release.

zwenjiang, 2021-12-09:
> @zwenjiang it seems curl installation is missing on the system. from the information provided I understand that you are trying to build ifm3d from source with these [instruction](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#source-dependencies). To build and install curl please follow these [instructions](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#curl). and after this you should be able to build xmlrpc-c with these [instructions](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#xmlrpc-c)
> 
> you might need to additionally pass -DBUILD_SHARED_LIBS=ON to cmake configure command of xmlrpc-c. We will update this in documentation in our next release.

> @zwenjiang it seems curl installation is missing on the system. from the information provided I understand that you are trying to build ifm3d from source with these [instruction](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#source-dependencies). To build and install curl please follow these [instructions](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#curl). and after this you should be able to build xmlrpc-c with these [instructions](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#xmlrpc-c)
> 
> you might need to additionally pass -DBUILD_SHARED_LIBS=ON to cmake configure command of xmlrpc-c. We will update this in documentation in our next release.

thanks for your replyment. I can compile it now.

closed 2021-12-09 by zwenjiang