ifm/ifm3d #338: ifm3d complie error about setting CMAKE_CXX_FLAGS

zwenjiang, 2021-12-09:
Hi, when I compile IFM3D with VS2019, I got the following compilation errors:

    正在创建库 F:/VPN/FeiruiRobots/myVision/trunk/IFM/ifm3d/ifm3d/build/modules/camera/src/libifm3d_camera/Release/ifm3d_camera.lib 和对象 F:/VPN/FeiruiRobots/myVision/trunk/IFM/ifm3d/ifm3d/build/modules/camera/src/libifm3d_camera/Release/ifm3d
  _camera.exp
  ifm3d_camera.vcxproj -> F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\camera\src\libifm3d_camera\Release\ifm3d_camera.dll
  Building Custom Rule F:/VPN/FeiruiRobots/myVision/trunk/IFM/ifm3d/ifm3d/modules/framegrabber/src/libifm3d_framegrabber/CMakeLists.txt
cl : 命令行 warning D9002: 忽略未知选项“-std=c++14” [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\src\libifm3d_framegrabber\ifm3d_framegrabber.vcxproj]
  byte_buffer.cpp
F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\include\ifm3d/fg/detail/byte_buffer.hpp(202,63): warning C4003: 类函数宏的调用“max”参数不足 [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabbe
r\src\libifm3d_framegrabber\ifm3d_framegrabber.vcxproj]
F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\include\ifm3d/fg/detail/byte_buffer.hpp(330,67): warning C4003: 类函数宏的调用“max”参数不足 [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabbe
r\src\libifm3d_framegrabber\ifm3d_framegrabber.vcxproj]
F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\modules\framegrabber\src\libifm3d_framegrabber\byte_buffer.cpp(92,44): warning C4003: 类函数宏的调用“max”参数不足 [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\src
\libifm3d_framegrabber\ifm3d_framegrabber.vcxproj]
**F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\modules\framegrabber\src\libifm3d_framegrabber\byte_buffer.cpp(92,44): error C2589: “(”:“::”右边的非法标记 [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\src\li
bifm3d_framegrabber\ifm3d_framegrabber.vcxproj]**
**F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\modules\framegrabber\src\libifm3d_framegrabber\byte_buffer.cpp(92): error C2062: 意外的类型“unknown-type” [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\src\l
ibifm3d_framegrabber\ifm3d_framegrabber.vcxproj]
F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\modules\framegrabber\src\libifm3d_framegrabber\byte_buffer.cpp(92,44): error C2059: 语法错误:“)” [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\src\libifm3d_
framegrabber\ifm3d_framegrabber.vcxproj]**
  frame_grabber.cpp
F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\include\ifm3d/fg/detail/byte_buffer.hpp(202,63): warning C4003: 类函数宏的调用“max”参数不足 [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabbe
r\src\libifm3d_framegrabber\ifm3d_framegrabber.vcxproj]
F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabber\include\ifm3d/fg/detail/byte_buffer.hpp(330,67): warning C4003: 类函数宏的调用“max”参数不足 [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\framegrabbe
r\src\libifm3d_framegrabber\ifm3d_framegrabber.vcxproj]
  Unknown compiler version - please run the configure tests and report the results
C:\Program Files\PCL 1.8.1\3rdParty\Boost\include\boost-1_64\boost/asio/detail/socket_types.hpp(24,1): fatal error C1189: #error:  WinSock.h has already been included [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\fra
megrabber\src\libifm3d_framegrabber\ifm3d_framegrabber.vcxproj]
  schema.cpp
  正在生成代码...
  Building Custom Rule F:/VPN/FeiruiRobots/myVision/trunk/IFM/ifm3d/ifm3d/modules/swupdater/src/libifm3d_swupdater/CMakeLists.txt
cl : 命令行 warning D9002: 忽略未知选项“-std=c++14” [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\swupdater\src\libifm3d_swupdater\ifm3d_swupdater.vcxproj]
  swupdater.cpp
**F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\modules\swupdater\src\libifm3d_swupdater\swupdater_impl.hpp(557,31): error C2593: “operator =”不明确 [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\swupdater\src\libifm3
d_swupdater\ifm3d_swupdater.vcxproj]**
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xstring(2965,42): message : 可能是“std::basic_string<char,std::char_traits<char>,std::allocator<char>> &std::basic_string<char,std::char_traits
<char>,std::allocator<char>>::operator =(std::initializer_list<_Elem>)” [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\swupdater\src\libifm3d_swupdater\ifm3d_swupdater.vcxproj]
          with
          [
              _Elem=char
          ]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xstring(2848,42): message : 或    “std::basic_string<char,std::char_traits<char>,std::allocator<char>> &std::basic_string<char,std::char_trai
ts<char>,std::allocator<char>>::operator =(std::basic_string<char,std::char_traits<char>,std::allocator<char>> &&) noexcept” [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\swupdater\src\libifm3d_swupdater\ifm3d_swupda
ter.vcxproj]
F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\modules\swupdater\src\libifm3d_swupdater\swupdater_impl.hpp(557,31): message : 尝试匹配参数列表“(std::string, nlohmann::basic_json<std::map,std::vector,std::string,bool,int64_t,uint64_t,double,
std::allocator,nlohmann::adl_serializer>)”时 [F:\VPN\FeiruiRobots\myVision\trunk\IFM\ifm3d\ifm3d\build\modules\swupdater\src\libifm3d_swupdater\ifm3d_swupdater.vcxproj]

inbangsa, 2021-12-09:
@zwenjiang May I know which version/branch/tag of ifm3d you are trying to compile with VS2019 ?

inbangsa, 2021-12-09:
Similar [issue](https://github.com/ifm/ifm3d/issues/284)

This is fixed in version 0.20.0, so you can upgrade it or can try a fix as explained in #284.

zwenjiang, 2021-12-09:
> pgrade it or can try a fi

Releases v0.18.0

zwenjiang, 2021-12-09:
> @zwenjiang May I know which branch/tag of ifm3d you are trying to compile with VS2019 ?

Releases v0.18.0

zwenjiang, 2021-12-09:
> Similar [issue](https://github.com/ifm/ifm3d/issues/284)
> 
> This is fixed in version 0.20.0, so you can upgrade it or can try a fix as explained in #284.

thank you, i try it

zwenjiang, 2021-12-10:
it's ok

closed 2021-12-10 by zwenjiang