ifm/ifm3d #340: Is there a UI available to configure O3R camera parameters? What is the maximum resolution supported for M03969?

mehranq, 2021-12-20:
Hi. I was wondering if there is a UI  available (like a web interface) to configure all of the available O3R camera parameters such as modes, filters, resolution, fov etc. or is it the case that O3R camera parameters can only be configured via json manipulation using the Get() and Set() commands in the library?

For the M03969, Is 224x172 the maximum resolution that is supported in 3D?

graugans, 2021-12-20:
Hey @mehranq we are working on a support for the O3R in the ifm Vision Assistant. Please get in touch with  [ifm support](mailto:support.robotics@ifm.com) it is maybe possible to get a preview of the Vision Assistant.

graugans, 2021-12-20:
Oh sorry I missed the question about the resolution. Yes the maximum resolution is 224x172. We have planned more heads with higher resolution. Please get in touch with the ifm sales team for more information when those will be available.

mehranq, 2021-12-20:
Thank you for your answer. About the support for the O3R in the ifm Vision Assistant, any estimate on when it will be officially available?

inbangsa, 2021-12-21:
@mehranq  as @graugans mention please reached out to ```support.robotics@ifm.com```  for the official release of O3R support in ifm Vision Assistant and they might  also help to get the  preview version of the same.

mehranq, 2021-12-21:
Will do that. Thank you.

closed 2021-12-21 by mehranq