ifm/ifm3d #341: can it work in macos

machineboy2018, 2021-12-21:
I compile the code in my macbook with m1 silicon successfully.But when I run the demo, it throw the error:
`libc++abi.dylib: terminating with uncaught exception of type ifm3d::error_t: Lib: Image chunk not found `
I want to know can it work in macos or I should do some other work to make it.

inbangsa, 2021-12-21:
@machineboy2018 we  offically donot support macos. 
but from error it seems there is mismatch between device and demo. can you please specify which demo you are trying to run and the device you are working with.

machineboy2018, 2021-12-21:
> @machineboy2018 we officially don't support macOS. but from error it seems there is mismatch between device and demo. can you please specify which demo you are trying to run and the device you are working with.

My device is O3D303.this is my code:
`int main(){
   auto _cam = ifm3d::Camera::MakeShared("192.168.0.69", 80, "");
   auto  _fg = std::make_shared<ifm3d::FrameGrabber>(_cam);
   auto _im = std::make_shared<ifm3d::ImageBuffer>(); 
    // Create pcic interface and initialize configuration
    ifm3d::PCICClient::Ptr pcic = std::make_shared<ifm3d::PCICClient>(_cam);
    while(1){
        if (!_fg->WaitForFrame(_im.get(), 1000)){
            std::cerr << "Timeout waiting for camera!" << std::endl;
            return 1;
        }
        else{
            cout << "success" << endl;
        }
    }
}`

I can ping the camera ip and can view it's data by windows 11 virtual machine.The macos forbiden user to use port 80, does this affect it's use.

It doesn't matter if it's difficult to solve it.I just think it will be cool to run in macos and can bring convenience to my work.

lola-masson, 2022-01-03:
Port 80 is the default XML_RPC port, which could explain your bug. Try switching it to something that your are allowed to use on macos.

Supporting macos is not on top of the priority list for us right now, but we will definitely announce it when it happens.

graugans, 2022-01-10:
Hey @machineboy2018  
> I can ping the camera ip and can view it's data by windows 11 virtual machine.The macos forbiden user to use port 80, does this affect it's use.
> 
> It doesn't matter if it's difficult to solve it.I just think it will be cool to run in macos and can bring convenience to my work.
I guess MacOS does not forbid to use a service which runs on port 80. Otherwise you would not be able to browse any http website on your MacOS device. On unix systems a regular user is typically not allowed to start a service which binds to a port below 1024, in case of ifm3d the XML-RPC service which binds to port 80 runs on the camera and your computer connects to this service as with any other http service. Related to the error message above the error is not related to the XML-RPC communication but related to the image data communication which is performed on a different port.

Can you please provide the information returned by the command line tool `ifm3d dump`?

closed 2022-10-23 by github-actions[bot]