ifm/ifm3d #342: STOI Error

rpangrazio, 2022-01-04:
I am getting an error with all commands of the system, including the ROS node. Every command I run gets a message of ifm3d error: stoi. The ROS node dies with a similar error. I have built from source for the tools, and I have installed precompiled versions. I am running on a Ubuntu 20.04 system.

rpangrazio, 2022-01-04:
Tried the 20.04 and 16.04 containers and same problem.

graugans, 2022-01-10:
Hey @rpangrazio, can you please describe the error in more details? As @tuxx0rz pointed out, can you also please share the device you are using O3D/O3X or O3R and the version of ifm3d you are using. In case you are running ROS please create an issue on the ROS driver.

rpangrazio, 2022-01-10:
Its an O3R.

So I rebuilt the 2004 container using main-next branch. No change. If I run any command of the ifm3d application that requires communication to the device it fails, e.g "ifm3d dump" or "ifm3d dump -ip 192.168.0.69". 

I can ping the sensor.

Please note that I am not using the ROS driver right now. I only mentioned it because I tried it. I am no longer trying to bring up the ROS system till I have this working.

graugans, 2022-01-10:
Can you please post the complete error message? Or is the error message simply 

```
error: stoi
```

rpangrazio, 2022-01-10:
ifm3d error: stoi

graugans, 2022-01-10:
That's strange, can you please check if `ifm3d` without any command does output the version?

```
ifm3d                    
ifm3d: version=0.90.0
```

If this also throws an error, what is the output of:
```
$ find /usr -name "*ifm3d*"
```

Can you also please provide an link to the Dockerfile you have used to build the Container?

rpangrazio, 2022-01-10:
```
ifm@48ce3b6989b0:~$ ifm3d -v
ifm3d: version=0.20.1
ifm@48ce3b6989b0:~$ ifm3d dump
ifm3d error: stoi
ifm@48ce3b6989b0:~$ which ifm3d
/usr/bin/ifm3d
ifm@48ce3b6989b0:~$ 
```

lola-masson, 2022-01-10:
```
ifm@48ce3b6989b0:~$ ifm3d -v
ifm3d: version=0.20.1
```
This version is not the one needed to interact with the O3R. After building from source from the branch o3r/main-next you should end up with something like 0.90+. I suspect something went wrong when you installed both from apt and from source. 
We have a docker container ready to use that you can directly pull, which should contain the most recent version of ifm3d and get you on your way:
https://github.com/ifm/ifm3d/pkgs/container/ifm3d

rpangrazio, 2022-01-10:
I am using the docker container.

lola-masson, 2022-01-10:
When I run that docker container, I get ifm3d version 0.92:
```
me@ME:~$  docker pull ghcr.io/ifm/ifm3d:latest-ubuntu
me@ME:~$ docker run -it ghcr.io/ifm/ifm3d:latest-ubuntu
ifm@e1363a064335:/$ ifm3d -v
ifm3d: version=0.92.0+39.47f850f
```

Did you pull that container recently?

rpangrazio, 2022-01-10:
I built the container today, but I am refollowing all the directions without regard to what I did before. Essentially starting from scratch, because I have a sinking suspicion I messed something up.

rpangrazio, 2022-01-10:
Pulled it from the repo instead of building locally worked. Thank you very much for the help.

lola-masson, 2022-01-10:
No worries at all :) Can you tell us which dockerfile you used to build before? There might be an error in the instructions there that I could correct.

rpangrazio, 2022-01-10:
All of the docker files in the repo under the docker folder. Tried them all. Used the python script to build them.

BigBoot, 2022-01-11:
Just for reference, the Dockerfiles for the o3r version are located in the docker/ifm-robotics folder, the others are all building the master version. So the one used for building the current version container would be this: https://github.com/ifm/ifm3d/blob/o3r/main-next/docker/ifm-robotics/docker/Dockerfile

closed 2022-10-22 by github-actions[bot]