ifm/ifm3d #344: Is the resolution on the O3R camera fixed?

mehranq, 2022-01-13:
Hi. Following is the configuration of an O3R camera with its 3D wire connected to port2 on the VPU. I see the resolution is set to 172x224. I understand that this is the maximum resolution for the camera. My question is, can we change the resolution to a lower setting? Or is it fixed to the max resolution and thus unchangeable through json manipulation?
`"port2": {
            "acquisition": {
                "exposureLong": 5000,
                "exposureShort": 400,
                "framerate": 10.0,
                "offset": 0.0,
                "version": {
                    "major": 0,
                    "minor": 0,
                    "patch": 0
                }
            },
            "data": {
                "algoDebugConfig": {},
                "availablePCICOutput": [],
                "pcicTCPPort": 50012
            },
            "info": {
                "device": "3101",
                "deviceTreeBinaryBlobOverlay": "001-irs2381c.dtbo",
                "features": {
                    "fov": {
                        "horizontal": 105,
                        "vertical": 78
                    },
                    "resolution": {
                        "height": 172,
                        "width": 224
                    },
                    "type": "3D"
                },
                "name": "",
                "partNumber": "M03969",
                "productionState": "AA",
                "sensor": "IRS2381C",
                "sensorID": "IRS2381C_105x78_4x2W_110x90_C7",
                "serialNumber": "000000000495",
                "vendor": "0001"
            },
            "mode": "standard_range4m",
            "processing": {
                "diParam": {
                    "anfFilterSizeDiv2": 2,
                    "enableDynamicSymmetry": true,
                    "enableStraylight": true,
                    "enableTemporalFilter": true,
                    "excessiveCorrectionThreshAmp": 0.3,
                    "excessiveCorrectionThreshDist": 0.08,
                    "maxDistNoise": 0.02,
                    "maxSymmetry": 0.4,
                    "medianSizeDiv2": 0,
                    "minAmplitude": 20.0,
                    "minReflectivity": 0.0,
                    "mixedPixelFilterMode": 1,
                    "mixedPixelThresholdRad": 0.15
                },
                "extrinsicHeadToUser": {
                    "rotX": 0.0,
                    "rotY": 0.0,
                    "rotZ": 0.0,
                    "transX": 0.0,
                    "transY": 0.0,
                    "transZ": 0.0
                },
                "version": {
                    "major": 0,
                    "minor": 0,
                    "patch": 0
                }
            },
            "state": "CONF"
        },`

lola-masson, 2022-01-13:
The resolution is fixed, as of right now we do not provides utilities for lowering the resolution.

mehranq, 2022-01-13:
Thank you for the clarification.

closed 2022-01-13 by mehranq

graugans, 2022-01-14:
Hey @mehranq , you can run a docker container on the VPu and downsample dat data on the VPU for your application.