ifm/ifm3d #345: Trouble using ifmO3r logger code

niallomahony93, 2022-01-27:
Hi, 

I am following the tutorial on https://ifm3d.com/logging2.html using the ifm03r pip module.

This is the code;
`import ifmO3r

import os
from ifmO3r.ifm3dTiny import Device, FrameGrabber, ImageBuffer, ImageLogger

IP = "192.168.0.69"
PORT = 50012

cam = Device(IP, PORT)
fg = FrameGrabber(cam)
im = ImageBuffer()
print("ok1")
cwd = os.getcwd()
data_path = os.path.join(cwd, 'data')

imgLogger = ImageLogger(cam, fg, im, data_path)

imgLogger.get_current_frame()
save_this = {
    "distance": "distance_frame1", 
    "amplitude": "amplitude_frame1", 
    "confidence": "confidence_frame1", 
    "x": "x_frame1", 
    "y": "y_frame1", 
    "z": "z_frame1",
    }
imgLogger.write_to_h5(save_this)
imgLogger.write_binary(save_this)
save_imgs = {
    "distance": "distance_frame1", 
    "amplitude": "amplitude_frame1", 
    "confidence": "confidence_frame1",
    }
imgLogger.write_png(save_imgs)

imgLogger.save_frame_as_pickle('frame1')
imgLogger.load_frame_from_pickle('data/frame1.p')`

When I use the logger module I get this error:

> Traceback (most recent call last):
  File "examples/o3r/viewer/ifm3dpy_logger.py", line 42, in <module>
    imgLogger.write_to_h5(save_this)
  File "/home/niall/anaconda3/envs/my_env/lib/python3.7/site-packages/ifmO3r/ifm3dTiny/logger.py", line 188, in write_to_h5
    confidence_img = self.imageBuffer.confidence_image()
  File "/home/niall/anaconda3/envs/my_env/lib/python3.7/site-packages/ifmO3r/ifm3dTiny/grabber.py", line 433, in confidence_image
    self.frame["image_width"]),
  File "<__array_function__ internals>", line 6, in reshape
  File "/home/niall/anaconda3/envs/my_env/lib/python3.7/site-packages/numpy/core/fromnumeric.py", line 299, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/home/niall/anaconda3/envs/my_env/lib/python3.7/site-packages/numpy/core/fromnumeric.py", line 58, in _wrapfunc
    return bound(*args, **kwds)
ValueError: cannot reshape array of size 77056 into shape (172,224)

It is like the camera is returning an array twice the size the logger is expecting. (77056 = 172*224*2) This is my camera config;

> "device": {
        "clock": {
            "currentTime": 1581091943025593920
        },
        "diagnostic": {
            "temperatures": [],
            "upTime": 1299000000000
        },
        "info": {
[... 263 lines cut ...]
                    "medianSizeDiv2": 0,
                    "minAmplitude": 20.0,
                    "minReflectivity": 0.0,
                    "mixedPixelFilterMode": 1,
                    "mixedPixelThresholdRad": 0.15
                },
                "extrinsicHeadToUser": {
                    "rotX": 0.0,
                    "rotY": 0.0,
                    "rotZ": 0.0,
                    "transX": 0.0,
                    "transY": 0.0,
                    "transZ": 0.0
                },
                "version": {
                    "major": 0,
                    "minor": 0,
                    "patch": 0
                }
            },
            "state": "RUN"
        }
    }
}
)

desengph, 2022-01-27:
Hi @niallomahony93 

The software code which you're referring to is “legacy code”. The Python package ifmo3r is unfortunately no longer supported.

Since the latest firmware releases the interface has changed, the effect is what you're seeing here. Part of this change has been the definition of the confidence image, which is now 16bit per pixel value instead of 8 bit.

We would like to suggest that you switch to our latest Python package – you can find it [here](https://pypi.org/project/ifm3dpy/).

For a more detailed description about the software features and the definition of all images, please follow these links.
[How to use the ifm3dpy library](http://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/examples/o3r/index.html)
[O3R images and filter: how to get started with your O3R](http://ifm3d.com/sphinx-doc/build/html/documentation/O3R/README.html) 

If you have any questions about the ifm3dpy or the O3R, please let us know.

j-friedl, 2022-07-29:
@desengph wrote:
> For a more detailed description about the software features and the definition of all images, please follow these links. [How to use the ifm3dpy library](http://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/examples/o3r/index.html)
However, there's no page ;)

lola-masson, 2022-07-29:
I am guessing this document has moved since the comment you mentioned. All the documentation is on [this](https://ifm3d.com/sphinx-doc/build/html/index.html) website.

closed 2022-10-22 by github-actions[bot]