ifm/ifm3d #347: Cannot Capture with O3R

UserFLO, 2022-02-07:
Hi ! I am using ifm VPU and O3R camera. I wrote a simple algorithm. 

from ifm3dpy import O3RCamera, ImageBuffer, FrameGrabber

# Initialize the objects
o3r = O3RCamera('192.168.0.69')
port='port2'
fg = FrameGrabber(o3r, pcic_port=50012)
im = ImageBuffer()

# Get a frame
if fg.wait_for_frame(im, 500)==False:
    raise ValueError #Exception('fg-timeout on ' + port + ' reached')

# Read the distance image and display a pixel in the center
dist = im.distance_image()
(width, height) = dist.shape
print(dist[width//2,height//2])__

When i turn on the VPU, the leds on port 0 and 2 don't turn on. Moreover, i cannot take a simple capture. The algorithm stops at : 
_if fg.wait_for_frame(im, 500)==False:
    raise ValueError #Exception('fg-timeout on ' + port + ' reached')_

Can you please help me ? 

I also check the configuration and with : 
_print(json.dumps(config,indent=4))_

I see that i have 2D camera on port 0 and 3D camera on port 2. 

about the state of each port, i see : CONF 

Thanks

graugans, 2022-02-07:
Hey @UserFLO , the heads are in config state by default. You have to change to `RUN` mode before any data can be received:

```json
'{ "ports": {"port2": {"state":"RUN"}}}' 
```
Given the above JSON fragment send to the O3R with `set` will enable the second camera.

UserFLO, 2022-02-09:
Thanks @graugans ! I have an other question. I would like to recover a 2D image. So i am using this algorithm in Python(itook the port0 and pcic_port = 50010) : 

_from ifm3dpy import O3RCamera, ImageBuffer, FrameGrabber

o3r = O3RCamera('192.168.0.69')
port='port0'
fg = FrameGrabber(o3r, pcic_port=50010)
im = ImageBuffer()

if fg.wait_for_frame(im, 500)==False:
    raise ValueError 

jpeg = im.jpeg_image()_

And i have this problem : cannot convert due to type or channel mistmatch 

Can you fix my issue ? 

Thank you

graugans, 2022-02-09:
Hey @UserFLO can you please post the output of the dump here?

UserFLO, 2022-02-09:
[image]

[image]

[image]

[image]

graugans, 2022-02-09:
Hey @UserFLO you also have to turn port0 into `RUN` mode:

```json
{ "ports": {"port0": {"state":"RUN"}}}
```

UserFLO, 2022-02-09:
I turn port0 into RUN ! But i think there is a conversion problem with the function _.jpeg_image()_ , because i obtain an array (108431,1) for my image. Do you know why ?

graugans, 2022-02-09:
I am not sure what you do expect. The array is the jpeg image data itself. If you wanttoi use it for example in OpenCV you can use it like this:

```python
cv2.imdecode(buf.jpeg_image(), cv2.IMREAD_UNCHANGED)
```

Please see the simple [python image viewer](cv2.imdecode(buf.jpeg_image(), cv2.IMREAD_UNCHANGED)
) for more details

UserFLO, 2022-05-02:
Hi @graugans i would like to know how i can save a PCL file with an O3R. 
First, i tried to use the xyz_image function but that doesn't work. 
Then, like i said i want to save my Point Cloud as PCL. 
Do you know how i could do that ?

Thanks you

desengph, 2022-05-03:
Hello @UserFLO,
most of the image data is handled as Numpy arrays internally. This also applies to the point cloud data. Please see this link for how to receive a point cloud image: [ifm3d.com point cloud docu](http://ifm3d.com/sphinx-doc/build/html/documentation/O3R/ProductsDescription/ImagesDescription/imagesDescription.html#point-cloud-xyz).

Please see the attached example for how to convert a Numpy array to a PLY pcd file format:
```python
def pcd_from_numpy_array(array, height, width, filename):
    # create an empty pcd file object
    pcd = open3d.geometry.PointCloud()

    # reshape data as open3d vector
    xyz = np.reshape(array, (3, height * width))
    xyz = np.transpose(xyz)
    pcd.points = open3d.utility.Vector3dVector(xyz)

    # save pcd file at cwd
    cwd = getcwd()
    saving_file = join(cwd, filename)
    try:
        open3d.io.write_point_cloud(saving_file, pcd)
    except Exception as e:
        status_logger.error(e)
        return False
    finally:
        return True
```
For a more complete example, check the content of this [link](http://www.open3d.org/docs/0.9.0/tutorial/Basic/working_with_numpy.html)

UserFLO, 2022-05-03:
Thank you for your answer. 
I'm not sure to use the right command line to get the point cloud. I tried with xyz_image but i think i'm doing something wrong : _im.xyz_image()_

desengph, 2022-05-03:
Hi @UserFLO,
the mentioned Python method `xyz_image()`is the correct one for getting the point cloud as a Numpy array.
See the details [here](http://ifm3d.com/sphinx-doc/build/html/api/ifm3dpy.ImageBuffer.html#ifm3dpy.ImageBuffer.xyz_image).

Please double-check the correct physical port - PCIC port mapping (i.e., check where a 3D imager is connected to in your JSON dump) and tat the respective imager has is in RUN state to receive data.

UserFLO, 2022-05-03:
[image]

I have this algorithm. But it returns me an error message : _Unsupported ifm3d::image type_

lola-masson, 2022-05-03:
It looks like you are grabbing frames from the wrong imager type.

Which version of ifm3dpy are you using? Could you send the current json configuration (without changing any physical connection)?

UserFLO, 2022-05-03:
[image]
[image]
[image]
[image]
[image]
@graugans

graugans, 2022-05-03:
Hey @UserFLO for us the JSON text would make things much easier. We can search and import the text. The screenshot is not that easy to read. If wou put 

&#96;&#96;&#96;json
{}
&#96;&#96;&#96;

UserFLO, 2022-05-03:
@graugans  i don't see where i can change the value. Or maybe i need to add these lines ?  If yes, where should I put them ?

UserFLO, 2022-05-03:
Like that ? Because nothing happens @graugans 
[image]

UserFLO, 2022-05-04:
Sorry ! I didn't understand what you were saying ! @graugans 

`''json 
{
    "device": {
        "clock": {
            "currentTime": 1581090730974955840
        },
        "diagnostic": {
            "temperatures": [],
            "upTime": 86000000000
        },
        "info": {
            "device": "0301",
            "deviceTreeBinaryBlob": "tegra186-quill-p3310-1000-c03-00-base.dtb",
            "features": {
                "imu": "BMI160"
            },
            "name": "",
            "partNumber": "M03975",
            "productionState": "AA",
            "serialNumber": "000201234185",
            "vendor": "0001"
        },
        "network": {
            "authorized_keys": "",
            "ipAddressConfig": 0,
            "macEth0": "00:04:4B:EA:9F:CE",
            "macEth1": "00:02:01:23:41:85",
            "networkSpeed": 100,
            "staticIPv4Address": "192.168.0.69",
            "staticIPv4Gateway": "192.168.0.201",
            "staticIPv4SubNetMask": "255.255.255.0",
            "useDHCP": false
        },
        "state": {
            "errorMessage": "",
            "errorNumber": ""
        },
        "swVersion": {
            "kernel": "4.9.140-l4t-r32.4+gc35f5eb9d1d9",
            "l4t": "r32.4.3",
            "os": "0.13.13-221",
            "schema": "v0.1.0",
            "swu": "0.15.12"
        }
    },
    "ports": {
        "port0": {
            "acquisition": {
                "framerate": 10.0,
                "version": {
                    "major": 0,
                    "minor": 0,
                    "patch": 0
                }
            },
            "data": {
                "algoDebugConfig": {},
                "availablePCICOutput": [],
                "pcicTCPPort": 50010
            },
            "info": {
                "device": "2301",
                "deviceTreeBinaryBlobOverlay": "001-ov9782.dtbo",
                "features": {
                    "fov": {
                        "horizontal": 127,
                        "vertical": 80
                    },
                    "resolution": {
                        "height": 800,
                        "width": 1280
                    },
                    "type": "2D"
                },
                "name": "",
                "partNumber": "M03933",
                "productionState": "AA",
                "sensor": "OV9782",
                "sensorID": "OV9782_127x80_noIllu_Csample",
                "serialNumber": "000000000611",
                "vendor": "0001"
            },
            "mode": "experimental_autoexposure2D",
            "processing": {
                "extrinsicHeadToUser": {
                    "rotX": 0.0,
                    "rotY": 0.0,
                    "rotZ": 0.0,
                    "transX": 0.0,
                    "transY": 0.0,
                    "transZ": 0.0
                },
                "version": {
                    "major": 0,
                    "minor": 0,
                    "patch": 0
                }
            },
            "state": "RUN"
        },
        "port2": {
            "acquisition": {
                "exposureLong": 5000,
                "exposureShort": 400,
                "framerate": 10.0,
                "offset": 0.0,
                "version": {
                    "major": 0,
                    "minor": 0,
                    "patch": 0
                }
            },
            "data": {
                "algoDebugConfig": {},
                "availablePCICOutput": [],
                "pcicTCPPort": 50012
            },
            "info": {
                "device": "3101",
                "deviceTreeBinaryBlobOverlay": "001-irs2381c.dtbo",
                "features": {
                    "fov": {
                        "horizontal": 60,
                        "vertical": 45
                    },
                    "resolution": {
                        "height": 172,
                        "width": 224
                    },
                    "type": "3D"
                },
                "name": "",
                "partNumber": "M03933",
                "productionState": "AA",
                "sensor": "IRS2381C",
                "sensorID": "IRS2381C_60x45_4x2W_60x45_C2",
                "serialNumber": "000000000611",
                "vendor": "0001"
            },
            "mode": "standard_range4m",
            "processing": {
                "diParam": {
                    "anfFilterSizeDiv2": 2,
                    "enableDynamicSymmetry": true,
                    "enableStraylight": true,
                    "enableTemporalFilter": true,
                    "excessiveCorrectionThreshAmp": 0.3,
                    "excessiveCorrectionThreshDist": 0.08,
                    "maxDistNoise": 0.02,
                    "maxSymmetry": 0.4,
                    "medianSizeDiv2": 0,
                    "minAmplitude": 20.0,
                    "minReflectivity": 0.0,
                    "mixedPixelFilterMode": 1,
                    "mixedPixelThresholdRad": 0.15
                },
                "extrinsicHeadToUser": {
                    "rotX": 0.0,
                    "rotY": 0.0,
                    "rotZ": 0.0,
                    "transX": 0.0,
                    "transY": 0.0,
                    "transZ": 0.0
                },
                "version": {
                    "major": 0,
                    "minor": 0,
                    "patch": 0
                }
            },
            "state": "RUN"
        }
    }
}

'''

lola-masson, 2022-05-04:
@UserFLO thanks for the configuration. It all looks fine to me. Can you check what version of the python package `ifm3dpy` you are using?

UserFLO, 2022-05-04:
@lola-masson 
```
pip show -f ifm3dpy

Name: ifm3dpy
Version: 0.92.0
Summary: Library for working with ifm pmd-based 3D ToF Cameras
Home-page: https://github.com/ifm/ifm3d
Author: ifm Robotics Perception
Author-email: support.robotics@ifm.com
License: Apache 2.0
Location: /home/fr543d/.local/lib/python3.8/site-packages
Requires: 
Required-by: 
Files:
  ifm3dpy-0.92.0.dist-info/INSTALLER
  ifm3dpy-0.92.0.dist-info/LICENSE
  ifm3dpy-0.92.0.dist-info/LICENSE-thirdparty
  ifm3dpy-0.92.0.dist-info/METADATA
  ifm3dpy-0.92.0.dist-info/RECORD
  ifm3dpy-0.92.0.dist-info/WHEEL
  ifm3dpy-0.92.0.dist-info/top_level.txt
  ifm3dpy.cpython-38-x86_64-linux-gnu.so
```

lola-masson, 2022-05-04:
This all looks fine to me. It might be easier to troubleshoot live. Could you send an email at support.robotics@ifm.com and we will setup a short video call to go through this together.

closed 2022-10-22 by github-actions[bot]