ifm/ifm3d #348: O3R Cyclic Mode Unstable Pointcloud

tuxx0rz, 2022-03-07:
I am trying out the cyclic_4m_2m_4m_2m mode on the O3R camera and VPU. After switching to the cyclic mode from the previous standard_range2m and setting up my extrinsic and filter setting I noticed the pointcloud is "bouncing" along the Z axis (Y Axis before my extrinsic settings) by approximately 30mm. I checked my pointcloud under standard_range2m and the Z axis values are correct. When checking the pointcloud under the standard_range4m the Z axis values are around 30mm too low. These checks were all done using the same camera extrinsic values as well as the same filter values and all using offset values of 0.0. The additional point data coming from the cyclic method would be very helpful to my application however this "bouncing"/ inconsistency between the 2m and 4m modes makes the data as of currently unusable in the cyclic mode. Am I thinking about / using this mode incorrectly or are there other factors at play? Below is a copy of my ifm3d dump. The camera in question currently is on Port 1.

Thanks,
Michael

{
  "device": {
    "clock": {
      "currentTime": 1646673383552559788
    },
    "diagnostic": {
      "temperatures": [],
      "upTime": 413503000000000
    },
    "info": {
      "device": "0301",
      "deviceTreeBinaryBlob": "tegra186-quill-p3310-1000-c03-00-base.dtb",
      "features": {},
      "name": "",
      "partNumber": "M03975",
      "productionState": "AA",
      "serialNumber": "00020142B0F6",
      "vendor": "0001"
    },
    "network": {
      "authorized_keys": "SSH KEYS REMOVED FOR POSTING",
      "ipAddressConfig": 0,
      "macEth0": "00:04:4B:EA:9F:7F",
      "macEth1": "00:02:01:42:B0:F6",
      "networkSpeed": 1000,
      "staticIPv4Address": "10.5.7.122",
      "staticIPv4Gateway": "10.5.7.1",
      "staticIPv4SubNetMask": "255.255.254.0",
      "useDHCP": false
    },
    "state": {
      "errorMessage": "",
      "errorNumber": ""
    },
    "swVersion": {
      "kernel": "4.9.140-l4t-r32.4+gc35f5eb9d1d9",
      "l4t": "r32.4.3",
      "os": "0.13.13-221",
      "schema": "v0.1.0",
      "swu": "0.15.12"
    }
  },
  "ports": {
    "port0": {
      "acquisition": {
        "exposureLong": 5000,
        "exposureShort": 400,
        "framerate": 10.0,
        "offset": 0.0,
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "data": {
        "algoDebugConfig": {},
        "availablePCICOutput": [],
        "pcicTCPPort": 50010
      },
      "info": {
        "device": "3101",
        "deviceTreeBinaryBlobOverlay": "001-irs2381c.dtbo",
        "features": {
          "fov": {
            "horizontal": 105,
            "vertical": 78
          },
          "resolution": {
            "height": 172,
            "width": 224
          },
          "type": "3D"
        },
        "name": "",
        "partNumber": "M03969",
        "productionState": "AA",
        "sensor": "IRS2381C",
        "sensorID": "IRS2381C_105x78_4x2W_110x90_C7",
        "serialNumber": "000000000521",
        "vendor": "0001"
      },
      "mode": "standard_range2m",
      "processing": {
        "diParam": {
          "anfFilterSizeDiv2": 3,
          "enableDynamicSymmetry": true,
          "enableStraylight": true,
          "enableTemporalFilter": true,
          "excessiveCorrectionThreshAmp": 0.3,
          "excessiveCorrectionThreshDist": 0.08,
          "maxDistNoise": 0.02,
          "maxSymmetry": 0.4,
          "medianSizeDiv2": 0,
          "minAmplitude": 10,
          "minReflectivity": 0.0,
          "mixedPixelFilterMode": 1,
          "mixedPixelThresholdRad": 0.15
        },
        "extrinsicHeadToUser": {
          "rotX": -1.561,
          "rotY": -0.02,
          "rotZ": -1.570796,
          "transX": -0.234,
          "transY": -1.45,
          "transZ": 1.1
        },
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "state": "RUN"
    },
    "port1": {
      "acquisition": {
        "exposureLong": 5000,
        "exposureShort": 400,
        "framerate": 10,
        "offset1": 0.0,
        "offset2": 0.0,
        "offset3": 0.0,
        "offset4": 0.0,
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "data": {
        "algoDebugConfig": {},
        "availablePCICOutput": [],
        "pcicTCPPort": 50011
      },
      "info": {
        "device": "3101",
        "deviceTreeBinaryBlobOverlay": "001-irs2381c.dtbo",
        "features": {
          "fov": {
            "horizontal": 105,
            "vertical": 78
          },
          "resolution": {
            "height": 172,
            "width": 224
          },
          "type": "3D"
        },
        "name": "",
        "partNumber": "M03969",
        "productionState": "AA",
        "sensor": "IRS2381C",
        "sensorID": "IRS2381C_105x78_4x2W_110x90_C7",
        "serialNumber": "000000000541",
        "vendor": "0001"
      },
      "mode": "cyclic_4m_2m_4m_2m",
      "processing": {
        "diParam": {
          "anfFilterSizeDiv2": 2,
          "enableDynamicSymmetry": false,
          "enableStraylight": false,
          "enableTemporalFilter": true,
          "excessiveCorrectionThreshAmp": 0.3,
          "excessiveCorrectionThreshDist": 0.08,
          "maxDistNoise": 0.01,
          "maxSymmetry": 0.4,
          "medianSizeDiv2": 1,
          "minAmplitude": 0.0,
          "minReflectivity": 0.0,
          "mixedPixelFilterMode": 1,
          "mixedPixelThresholdRad": 0.25
        },
        "extrinsicHeadToUser": {
          "rotX": -1.6,
          "rotY": -0.04,
          "rotZ": -1.570796,
          "transX": -0.226,
          "transY": -1.425,
          "transZ": 0.207
        },
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "state": "RUN"
    },
    "port2": {
      "acquisition": {
        "framerate": 10.0,
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "data": {
        "algoDebugConfig": {},
        "availablePCICOutput": [],
        "pcicTCPPort": 50012
      },
      "info": {
        "device": "2301",
        "deviceTreeBinaryBlobOverlay": "001-ov9782.dtbo",
        "features": {
          "fov": {
            "horizontal": 127,
            "vertical": 80
          },
          "resolution": {
            "height": 800,
            "width": 1280
          },
          "type": "2D"
        },
        "name": "",
        "partNumber": "M03969",
        "productionState": "AA",
        "sensor": "OV9782",
        "sensorID": "OV9782_127x80_noIllu_Csample",
        "serialNumber": "000000000521",
        "vendor": "0001"
      },
      "mode": "experimental_autoexposure2D",
      "processing": {
        "extrinsicHeadToUser": {
          "rotX": 0.0,
          "rotY": 0.0,
          "rotZ": 0.0,
          "transX": 0.0,
          "transY": 0.0,
          "transZ": 0.0
        },
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "state": "RUN"
    },
    "port3": {
      "acquisition": {
        "framerate": 10.0,
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "data": {
        "algoDebugConfig": {},
        "availablePCICOutput": [],
        "pcicTCPPort": 50013
      },
      "info": {
        "device": "2301",
        "deviceTreeBinaryBlobOverlay": "001-ov9782.dtbo",
        "features": {
          "fov": {
            "horizontal": 127,
            "vertical": 80
          },
          "resolution": {
            "height": 800,
            "width": 1280
          },
          "type": "2D"
        },
        "name": "",
        "partNumber": "M03969",
        "productionState": "AA",
        "sensor": "OV9782",
        "sensorID": "OV9782_127x80_noIllu_Csample",
        "serialNumber": "000000000541",
        "vendor": "0001"
      },
      "mode": "experimental_autoexposure2D",
      "processing": {
        "extrinsicHeadToUser": {
          "rotX": 0.0,
          "rotY": 0.0,
          "rotZ": 0.0,
          "transX": 0.0,
          "transY": 0.0,
          "transZ": 0.0
        },
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "state": "RUN"
    }
  }
}

lola-masson, 2022-03-08:
Hi Michael. 
It is possible that you are seeing a difference between the 2 and 4m in cyclic mode due to MPI (multi-path interference). The range being different for the two modes, the impact of this artifact (light bouncing against a wall or the floor for instance before reflecting off the object of interest) can be different.
Are you seeing this difference when switching between 2 and 4m modes, without using the cyclic mode?

tuxx0rz, 2022-03-08:
Yes I did notice a difference when switching between the 2m and 4m modes. To get the same x/y/z coords between the 2m and 4m mode I have to adjust the transY on port 1 by approximately 0.03 however the transY is the only extrinsic parameter that needs to be modified between the 2 modes.

lola-masson, 2022-03-08:
Could you send a recording of the scene you are looking at? If this needs to be shared privately you can email me at lola.masson@ifm.com. I can have a look at rosbags if you have that available, or a video recording can be a good start.
The MPI would most likely appear in a scene where there is a corner between a floor and a wall or a floor and an object, or something like that.

closed 2022-04-29 by tuxx0rz