ifm/ifm3d #350: Problems when using lfm3dpy

LarsBuck, 2022-03-09:
Hello,

so i was experiencing some issues when using the O3D300 Sensor with python ifm3dpy library.
My Setup:

Ubuntu 20.04 LTS
5.13.0-30-generic
Python 3.8.10
ifm3dpy 0.93.0
ifm3d 0.18.0
installed with snap
camera firmware 1.71.xx

The problem:

I was using the example code given on your docs [HERE](https://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/examples/basic_lib_usage/first_steps_ifm3dpy.html#receive-an-image]) and experienced some errors regarding the .get() and other methods.

importlib._bootstrap.Error: Unknown error 100001: XMLRPC method not found (D-Bus error 16)

Have you experienced the same error before and can help me out with a possible fix ? If you could provide any further examples that would be helpful.

inbangsa, 2022-03-09:
Hello @LarsBuck ,

The document you referred is written for O3R device, For O3D/O3X  camera please refer to the [Usage Example](https://github.com/ifm/ifm3d/blob/master/doc/python.md#usage) 

you mentioned in your issue that you have O3D sensor and not camera. if its sensor then you might need to create a default application on device and make it active to grab the data.

graugans, 2022-03-09:
Hey @LarsBuck please check those examples [example-creating-new-applications](https://github.com/ifm/ifm3d/blob/master/doc/cmdline.md#example-creating-new-applications)

LarsBuck, 2022-03-09:
Hi,

so i just took a look at the sensor devices via `$ ifm3d ls` and got the following output

 {
   "Active": true,
   "Description": "",
   "Id": 1571000710,
   "Index": 1,
   "Name": "Advanced application"
 }
]

So i guess there is an active application for my sensor. How exactly do I grab the data from the Sensor ? I ran the example code provided by @inbangsa which will return the amplitude image in form of a numpy array. I then tried to visualize this with the opencv imshow() method, but it will only show a fully black image.

inbangsa, 2022-03-09:
@LarsBuckcan you help me understand the fully black image, is it all the values in the numpy array are zero OR it might be the values are near to zero and that might cause image to appear black.

LarsBuck, 2022-03-09:
The amplitude image holds values out of range 

[[1174 1091 1151 ... 591 611 629]
[1121 1058 1130 ... 755 718 750]
[ 965 974 1161 ... 990 968 989]
...
[ 626 638 754 ... 862 746 676]
[ 593 581 710 ... 818 715 668]
[ 845 785 766 ... 778 719 666]]

I also tried to normalize them into range of 0...255 but without success.

[image]

inbangsa, 2022-03-09:
@LarsBuck if I look closely I can see the box in Image. The issue here is not with ```ifm3dpy``` but its with display. 

for amplitude image of O3D3XX each pixel is stored in ```uint16_t```

* for Display you can normalize image data with min max of the image. 
OR
* Just to see data you can multiply the numpy array with a constant value so that overall brightness will increase.

inbangsa, 2022-03-09:
@LarsBuck [this issue](https://github.com/ifm/ifm3d/issues/307) looks very similar. Please refer to discussion for scaling and Display of the amplitude image

inbangsa, 2022-03-09:
@LarsBuck if problem is resolve can we close this issue.

closed 2022-03-09 by LarsBuck