ifm/ifm3d #352: Some issue about PDS

GoodTekken, 2022-03-22:
1, PSD needs to send several counts of trigger commands until detecting the plastic pallet in the same environment.
The fail hints "status: -1036 (Not enough sides to construct pockets)"
[image]

The success hints "status: 0 (OK)"
elapsed_time: 1.33831512928
confidence: 0.926914989948
pallet_x: 1.52500009537
pallet_y: -0.0243749693036
pallet_z: 0.00750002264977
left_pocket_x: 1.54400002956
left_pocket_y: 0.204375043511
left_pocket_z: 0.00750002264977
right_pocket_x:1.50600004196
right_pocket_y: -0.253124982119
right_pocket_z: 0.00750002264977
roll: 0.0
pitch: 0.0
yaw: -0.078842908144

[image]

2, The point cloud distorted (flat surfaces bent, ...)， with attachment.
（I have set           
"EnableRectificationAmplitudeImage": "true",
"EnableRectificationDistanceImage": "true",)
As for the current setup, I pasted out camera dump into this Pastebin:
([https://pastebin.com/TdN7d17B])

3, If I want to make the algorithm of PDS by myself， Do you have any suggestions or conduction?
(https://www.ifm.com/us/en/us/robotics/software-solutions/pds/integrating-pds/integrating-pds)

GreNait, 2022-03-22:
Hey @GoodTekken,

PDS is a special software and has nothing to do with the ifm3d library. I will write you a PM with my work e-mail, so we can take this discussion off GitHub.

GoodTekken, 2022-03-23:
Okay, It's very nice of you to contact me.Thank you.

closed 2022-03-23 by GoodTekken