ifm/ifm3d #353: Problem with glog

GoodTekken, 2022-04-29:
Hello， I have the same issue with [ERROR: something wrong with flag 'logtostderr' in file 'src/logging.cc'.  One possibility: file 'src/logging.cc' is being linked both statically and dynamically into this executable.](https://github.com/ifm/ifm3d/issues/40).

But  I cannot pass the compile even if follow the step. 

> ifm3d --version

> ifm3d: version=0.18.0

> uname -a

> Linux JT 5.13.0-40-generic #45~20.04.1-Ubuntu SMP Mon Apr 4 09:38:31 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

The version of QT is 5.14.2，and I found that if I just use the <glog/logging.h> individual it can pass complie. But when I add the [auto cam = ifm3d::Camera::MakeShard()](https://github.com/ifm/ifm3d/blob/master/doc/basic_usage.md) it will be  error.

My ubuntu is 20.04,  and ifm3d is 0.18.0, So I doubt that if anything compatibility with the ifm3d library.

lola-masson, 2022-05-02:
Could you tell me how you installed ifm3d? Did you build from source or use apt?

GoodTekken, 2022-05-02:
Use apt below:

$ sudo apt-get update
$ sudo apt-get install ifm3d-camera \
                       ifm3d-framegrabber \
                       ifm3d-swupdater \
                       ifm3d-image \
                       ifm3d-opencv \
                       ifm3d-pcicclient \
                       ifm3d-tools \
                       ifm3d-python3 \
                       ifm3d-pcl-viewer \

graugans, 2022-05-02:
@GoodTekken ,are you using glog from the Ubuntu apt-repository or do you have a local installation from source or do you have glog included into your code base. There is an issue if you mix the versions of glog.

GoodTekken, 2022-05-03:
Many Thanks. @lola-masson @graugans ,It can work now when I specify the library below.

`#glog
win32:CONFIG(release, debug|release): LIBS += -L$$PWD/../../../../lib/x86_64-linux-gnu/release/ -lglog
else:win32:CONFIG(debug, debug|release): LIBS += -L$$PWD/../../../../lib/x86_64-linux-gnu/debug/ -lglog
else:unix: LIBS += -L$$PWD/../../../../lib/x86_64-linux-gnu/ -lglog

INCLUDEPATH += $$PWD/../../../../lib/x86_64-linux-gnu
DEPENDPATH += $$PWD/../../../../lib/x86_64-linux-gnu`

closed 2022-05-04 by GoodTekken