ifm/ifm3d #354: Timeout waiting for camera

GoodTekken, 2022-05-11:
Hello,sir.
         Today I found that camera will take several seconds to get an image use the code under:
```
        fg->SWTrigger(); //TRIGGER
        if (! fg->WaitForFrame(im.get(), 0))
        {
          std::cerr << "Timeout waiting for camera!" << std::endl;
          return -1;
        }
```
     And I notice that when camera temperature up to 50℃ will cause that happen. It it can word normally when the temperature around 30℃.
  
   Do you have any advice?

GoodTekken, 2022-05-11:
When Temperature 36.7℃，the elapsed_time:18435ms

graugans, 2022-05-11:
Hey @GoodTekken , what type of camera are you using? I guess it is an O3D3xx am I right?
The general recommendation is to mount the camera in a way that there is a good heat dissipation provided.

GoodTekken, 2022-05-11:
Yes,sir. The camera is O3DP03.
And Ping is okay.
```
jt@JT:~$ ping 192.168.0.69
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.
64 bytes from 192.168.0.69: icmp_seq=1 ttl=64 time=0.276 ms
64 bytes from 192.168.0.69: icmp_seq=2 ttl=64 time=0.359 ms
64 bytes from 192.168.0.69: icmp_seq=3 ttl=64 time=0.365 ms
64 bytes from 192.168.0.69: icmp_seq=4 ttl=64 time=0.405 ms
```
Now I switch the camera to string mode and use the order with"ifm",It hints "**Timeout waiting for camera!**" too.

```
[ INFO] [1652252488.667637285]: Loading nodelet /ifm3d_ros_examples/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet with the following remappings:
[ INFO] [1652252488.671071466]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1652252488.677003864]: Initializing nodelet with 4 worker threads.
[ INFO] [1652252488.692297637]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] is now available.
[ INFO] [1652252488.756652292]: IP default: 192.168.0.69, current 192.168.0.69
[ INFO] [1652252488.757925027]: pcic port check: current 50010, default 50010
[ INFO] [1652252488.849718817]: Running dtors...
[ INFO] [1652252488.849813273]: Initializing camera...
[ INFO] [1652252489.866697382]: Initializing framegrabber...
[ INFO] [1652252489.867918164]: Nodelet arguments: 16, 50010
[ INFO] [1652252489.868028264]: Initializing image buffer...
[ WARN] [1652252490.369207744]: Timeout waiting for camera!
[ WARN] [1652252490.869767394]: Timeout waiting for camera!

```

GoodTekken, 2022-05-11:
The mount is same as preview. And the camera can run well a few days ago.
What can make the camera timeout? Or XMLRPC ?

graugans, 2022-05-11:
Okay, it looks like you are using the O3DP03 with the standard ROS node. Typically the O3DP03 comes with its own interface which configures the camera to software trigger. As in your example above one has to trigger the camera. As far as I know the standard ROS node expects the camera to be configured in continuous mode. This means it times out because no data is received because no one triggers the camera.

Can you please describe what is your goal and what data you are looking for?

GoodTekken, 2022-05-11:
Yes, I have set the camera in continuous mode, but still "TimeOut".

I want to get the point cloud from the camera and make a test for pallet detection.

GoodTekken, 2022-05-11:
How odd! I can get the data from the camera before. No mather triggered mode or continuous mode.

graugans, 2022-05-11:
Can you please grab the trace logs from the camera?

```
$ ifm3d trace
````

GoodTekken, 2022-05-11:
No , it cannot.

```
jt@JT:~$ ifm3d trace
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?
jt@JT:~$ ping 192.168.0.69
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.
64 bytes from 192.168.0.69: icmp_seq=1 ttl=64 time=0.243 ms
64 bytes from 192.168.0.69: icmp_seq=2 ttl=64 time=0.341 ms
```

graugans, 2022-05-11:
Please try
````
$ ifm3d trace --limit 100
````

There is a known bug in the libxmlrpc when it comes to larger chunks of data

GoodTekken, 2022-05-11:
```
jt@JT:~$ ifm3d trace --limit 100
ifm3d error: too many positional options have been specified on the command line
jt@JT:~$ ifm3d trace --limit 1
ifm3d error: too many positional options have been specified on the command line
jt@JT:~$ ifm3d trace --limit 2
ifm3d error: too many positional options have been specified on the command line
jt@JT:~$ ifm3d trace --limit 1000
ifm3d error: too many positional options have been specified on the command line
jt@JT:~$ ifm3d trace --limit 100
ifm3d error: too many positional options have been specified on the command line
```

graugans, 2022-05-11:
Sorry there was a typo:

````
$ ifm3d trace --limit=100
````

GoodTekken, 2022-05-11:
$ ifm3d trace --limit=100
https://pastebin.com/H2ygapN6[$ ifm3d trace --limit=100](url)

graugans, 2022-05-11:
```
T: 366.4476, P: 375,  C: WARN,  S: iom,  M: DIAGC_DBUS_EVENT_UB_UNDERVOLT
T: 366.4504, P: 375,  C: WARN,  S: iom,  M: DIAGC_DBUS_EVENT_VLED_UNDERVOLT
T: 366.4521, P: 375,  C: WARN,  S: iom,  M: DIAGC_DBUS_EVENT_UB_UNDERVOLT
T: 395.5820, P: 375,  C: WARN,  S: iom,  M: DIAGC_DBUS_EVENT_VLED_UNDERVOLT
````
Indicates your power supply is not able to provide the specified current and voltages. Please check your power-supply

GoodTekken, 2022-05-11:
It can get the Image suddenly , the trace below:
[can get the image](https://pastebin.com/x1Q3fHTd)

graugans, 2022-05-11:
May I suggest to use a power-supply for testing, to get it out of the loop.

GoodTekken, 2022-08-10:
Thank you, sir. And sorry for the late reply.
when I use the power supply, It can work quite well.

closed 2022-08-10 by GoodTekken