ifm/ifm3d #355: Multiple failures on 'make check' (Ubuntu 20.04 build from source)

KtGunn, 2022-07-11:
I am building from source on Ubuntu 20.04. When running make check I get multiple failures:

[----------] Global test environment tear-down
[==========] 21 tests from 2 test suites ran. (87 ms total)
[  PASSED  ] 11 tests.
[  FAILED  ] 10 tests, listed below:
[  FAILED  ] StlImageBuffer.MoveCtor
[  FAILED  ] StlImageBuffer.MoveAssignmentOperator
[  FAILED  ] StlImageBuffer.CopyCtor
[  FAILED  ] StlImageBuffer.CopyAssignmentOperator
[  FAILED  ] StlImageBuffer.References
[  FAILED  ] StlImageBuffer.ComputeCartesian
[  FAILED  ] StlImageBuffer.TimeStamp
[  FAILED  ] StlImageBuffer.IlluTemp
[  FAILED  ] StlImageBuffer.DistanceNoiseImage
[  FAILED  ] StlImage.setTo

Most of the error messages in the test of StlImageBuffer are:
C++ exception with description "Lib: The detected device is not supported by the library" thrown in the test body.

I am able to ping the OVP800 at 192.168.0.69, the default IP, although the message with the last failed StlImageBuffer test, DistanceNoiseImage test is:
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.

Information about my environment:
Linux Ubuntu-Precision-5560-1 5.14.0-1044-oem #49-Ubuntu SMP Mon Jun 27 12:26:11 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 20.04.4 LTS
IFM3D version:
v0.93.0+0.0042452*v0.93.0*0*93*0**0*0042452

Can you please advice me how to proceed to complete the install successfully?
Thank you.

graugans, 2022-07-12:
Hey @KtGunn,

maybe the firmware on your device is outdated. Can you please check this:

```
$ curl http://192.168.0.69/api/rpc/v1/com.ifm.efector/?method=getSWVersion
```
I do get a response like this:
```
<div style="border: 1px solid; padding: 2px"><b>Response</b><br>{
    "error": null,
    "id": null,
    "result": {
        "IFM_Software": "0.16.3-643",
        "Linux": "Linux version 4.9.140-l4t-r32.4+ga89de209c6a7 (oe-user@oe-host) ",
        "Main_Application": "1.9.0"
    }
}
</div><br><form method='post'>id: <input name='id' type='text' value='1'><br><i>(JSON-RPC id, will directly be copied to answer-msg)</i><br>
method: <input name='method' type='text'><br><i>(e.g. org.freedesktop.DBus.Introspectable.Introspect, org.freedesktop.DBus.ListNames)</i><br>params: <input name='params' type='text' value='[]'><br><i>(The bridge now support minimal type-convertion ... //FIXME: atm there are still some parts missing for non-primitiv data ...)</i><br>
<input type='submit'><br></form>
<div>methods:<ul><li>com.ifm.efector.getParameter(s parameterName) -> (is) status, s parameterValue</li><li>com.ifm.efector.getAllParameters() -> (is) status, a{sv} parameterMap</li><li>com.ifm.efector.getClientCompatibilityList() -> (is) status, a{s} compatibilityList</li><li>com.ifm.efector.getApplicationList() -> (is) status, aa{sv} applicationList</li><li>com.ifm.efector.get(a{v} jsonPointers) -> (is) status, s jsonConfiguration</li><li>com.ifm.efector.set(s jsonConfiguration) -> (is) status, s result</li><li>com.ifm.efector.remove(s jsonObject) -> (is) status, s result</li><li>com.ifm.efector.reset(s jsonObject) -> (is) status, s result</li><li>com.ifm.efector.getSchema() -> (is) status, s jsonSchema</li><li>com.ifm.efector.getInit() -> (is) status, s jsonInitConfig</li><li>com.ifm.efector.saveInit() -> (is) status</li><li>com.ifm.efector.getSWVersion() -> (is) status, a{sv} versionMap</li><li>com.ifm.efector.applyNetworkConfiguration() -> (is) status</li><li>com.ifm.efector.reboot() -> (is) status</li><li>com.ifm.efector.factoryReset(b keepNetworkSettings) -> (is) status</li><li>com.ifm.efector.getTraceLogs(i nLogs) -> (is) status, as logs</li><li>org.freedesktop.DBus.Properties.Get(s interface_name, s property_name) -> v value</li><li>org.freedesktop.DBus.Properties.Set(s interface_name, s property_name, v value) -> </li><li>org.freedesktop.DBus.Properties.GetAll(s interface_name) -> a{sv} values</li><li>org.freedesktop.DBus.Introspectable.Introspect() -> s xml_data</li><li>org.freedesktop.DBus.Peer.Ping() -> </li><li>org.freedesktop.DBus.Peer.GetMachineId() -> s machine_uuid</li></ul></div>
```

lola-masson, 2022-07-13:
Hi @KtGunn. After double checking (no pun intended), the `make check` step is not fully functional for the O3R platform yet. 
Please make sure you follow the build instructions [here](https://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/installation_instructions/source_build.html), using a tagged version (latest is [0.93.0](https://github.com/ifm/ifm3d/tree/v0.93.0)) from github. 

Regarding the firmware, 14.23 is the latest version so you should be good. For future updates, the firmware can be downloaded on ifm.com (you will need to create an account and then check the product page for your VPU and go to the downloads tab). Release notes and instructions are on the [developer portal](https://ifm3d.com/sphinx-doc/build/html/documentation/O3R/ProductsDescription/Firmware_0.14.x_Release_Notes.html)

closed 2022-10-22 by github-actions[bot]