ifm/ifm3d #357: Build the ifm3d from source

vdorsi, 2022-07-22:
Hello all !

I am trying to to build the ifm3d library from source using the version 0.93.0
`git clone -b o3r/main-next https://github.com/ifm/ifm3d.git) `

and the following as described in the O3R developper guide :
```
$ mkdir build
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_MODULE_OPENCV=ON -DBUILD_MODULE_PCICCLIENT=ON -DBUILD_MODULE_IMAGE=ON  ..
$ cmake --build .
$ sudo cmake --build . --target install
```

In the documentation it is described the "camera" and "stlimage" modules will be builded by default. But in my case it is not, I have only the following packages : device, framegraber, pcicclient, pybind11, swupdater and tools.

I have the OPV800 product and I would like to use the ifm3d-ros2 library to develop the project.
So the result when I want to build the ifm3d-ros2 library, I got the following error message : 
- ifm3d could not find component: camera
- ifm3d could not find component: stlimage
[image]

I am working on the virtual machine wih Ubuntu 20.04.
Is someone have the same issue ?

Thank you for your support.

vdorsi, 2022-07-25:
Hello again !

I have resolved the issue ! 
The issue was from the `git clone -b o3r/main-next https://github.com/ifm/ifm3d.git` command ! 
It is not cloning all the sources files...  I have downloaded manually the tar.gz archive from the repository and everything is working now...

graugans, 2022-07-25:
Hej @vdorsi thabnks for the input. We will check this and maybe update the documentation if required.

vdorsi, 2022-07-25:
Hi !
I am now trying to test the example application of getting a stream from the O3R camera.
The script if "ifm3d_ros2/camera_standalone" and I am getting the following error : 

"Timeout waiting for camera" after starting the publishing loop.
[image]

Do you know what can be the source of this timeout ? If I running the IfmVisionAssistant, I have all the stream for the both cameras I have on the OVP800.

Thanks

desengph, 2022-07-25:
Hi @vdorsi 
it seems to me that you're trying to get images from a camera head which is not in `RUN` state. Can you please double-check that the camera stream, which you're trying to use, is available?
This can either be done via the ifm Vision Assistant (as you have done before), via the dump and config methods, or the Soft-On Soft-Off helpers.

Additionally, can you please try to run the launch files as mentioned [here](https://github.com/ifm/ifm3d-ros/tree/master/ifm3d_ros_examples#nodelet-launch-structure). 

I hope this helps. If you have any additional questions, can you please open an issue on the respective GitHub repositories for ROS.

closed 2022-07-25 by vdorsi

lola-masson, 2022-07-25:
Hi @vdorsi,

A precision for the build:
When you run `git clone -b o3r/main-next https://github.com/ifm/ifm3d.git`, it does **not** automatically gets you to the source for version 0.93.0. You need to checkout the `v0.93.0` branch before you follow the build instructions. The branch `o3r/main-next` contains the latest changes for the library and is a work in progress. We do not recommend building using that branch.
Can you check the result of `ifm3d --version` and `which ifm3d`? I just want to make sure you are using the right version of the library.

vdorsi, 2022-07-25:
Hi @lola-masson ,

Thank you for your answer.
The current version of the ifm3d is 0.93.0+114.a83ee75
And the location is /usr/bin/ifm3d
[image]