ifm/ifm3d #361: Image retrieval error - Python library

nikhilnagaraj, 2022-08-01:
Hi,

I am following the tutorial [here](https://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/examples/o3r/getting_data/getting_data.html) and I am having trouble retrieving the image.

I encounter this error: 
[image].

I am able to connect to the VPU (data packets are received when using ping). The python library version is `0.93` and the firmware version is `0.14.23` . Both the ports (2D and 3D) have been set to `RUN`.

Is there something I am missing here?

j-friedl, 2022-08-01:
Are you using the exact demo code? Also, could you share `ifm3d dump`?

nikhilnagaraj, 2022-08-01:
Yes, the exact demo code. 

```
{
    "device": {
        "clock": {
            "currentTime": 1581090672946101312
        },
        "diagnostic": {
            "temperatures": [
                {
                    "entity": "PLL-therm",
                    "value": 36.0
                },
                {
                    "entity": "MCPU-therm",
                    "value": 36.0
                },
                {
                    "entity": "PMIC-Die",
                    "value": 100.0
                },
                {
                    "entity": "Tboard_tegra",
                    "value": 33.0
                },
                {
                    "entity": "GPU-therm",
                    "value": 32.5
                },
                {
                    "entity": "BCPU-therm",
                    "value": 36.0
                },
                {
                    "entity": "thermal-fan-est",
                    "value": 34.29999923706055
                },
                {
                    "entity": "Tdiode_tegra",
                    "value": 31.75
                },
                {
                    "entity": "port2",
                    "value": 0
                }
            ],
            "upTime": 28000000000
        },
        "info": {
            "description": "",
            "device": "0302",
            "deviceTreeBinaryBlob": "tegra186-quill-p3310-1000-c03-00-base.dtb",
            "guiSettings": "",
            "name": "",
            "partNumber": "OVP800",
            "productionState": "AA",
            "serialNumber": "000018384491",
            "vendor": "0001"
        },
        "log": {
            "level": "warn",
            "storage": "volatile"
        },
        "network": {
            "authorized_keys": "",
            "interfaces": {
                "eth0": {
                    "ipAddressConfig": 0,
                    "ipv4": {
[... 308 lines cut ...]
                    "medianSizeDiv2": 0,
                    "minAmplitude": 20.0,
                    "minReflectivity": 2.0,
                    "mixedPixelFilterMode": 1,
                    "mixedPixelThresholdRad": 0.05
                },
                "extrinsicHeadToUser": {
                    "rotX": 0.0,
                    "rotY": 0.0,
                    "rotZ": 0.0,
                    "transX": 0.0,
                    "transY": 0.0,
                    "transZ": 0.0
                },
                "version": {
                    "major": 0,
                    "minor": 11,
                    "patch": 12
                }
            },
            "state": "CONF"
        }
    }
}
```

nikhilnagaraj, 2022-08-01:
Yes, the exact demo code. Here's the dump:

```
{
    "device": {
        "clock": {
            "currentTime": 1581091029598348768
        },
        "diagnostic": {
            "temperatures": [
                {
                    "entity": "PLL-therm",
                    "value": 36.0
                },
                {
                    "entity": "MCPU-therm",
                    "value": 36.0
                },
                {
                    "entity": "PMIC-Die",
                    "value": 100.0
                },
                {
                    "entity": "Tboard_tegra",
                    "value": 33.0
                },
                {
                    "entity": "GPU-therm",
                    "value": 33.0
                },
                {
                    "entity": "BCPU-therm",
                    "value": 36.0
                },
                {
                    "entity": "thermal-fan-est",
                    "value": 34.79999923706055
                },
                {
                    "entity": "Tdiode_tegra",
                    "value": 32.25
                },
                {
                    "entity": "port2",
                    "value": 0
                }
            ],
            "upTime": 385000000000
        },
        "info": {
            "description": "",
            "device": "0302",
            "deviceTreeBinaryBlob": "tegra186-quill-p3310-1000-c03-00-base.dtb",
            "guiSettings": "",
            "name": "",
            "partNumber": "OVP800",
            "productionState": "AA",
            "serialNumber": "000018384491",
            "vendor": "0001"
        },
        "log": {
            "level": "warn",
            "storage": "volatile"
        },
        "network": {
            "authorized_keys": "",
            "interfaces": {
                "eth0": {
                    "ipAddressConfig": 0,
                    "ipv4": {
[... 308 lines cut ...]
                    "medianSizeDiv2": 0,
                    "minAmplitude": 20.0,
                    "minReflectivity": 2.0,
                    "mixedPixelFilterMode": 1,
                    "mixedPixelThresholdRad": 0.05
                },
                "extrinsicHeadToUser": {
                    "rotX": 0.0,
                    "rotY": 0.0,
                    "rotZ": 0.0,
                    "transX": 0.0,
                    "transY": 0.0,
                    "transZ": 0.0
                },
                "version": {
                    "major": 0,
                    "minor": 11,
                    "patch": 12
                }
            },
            "state": "RUN"
        }
    }
}
```

inbangsa, 2022-08-01:
@nikhilnagaraj Can you please specify the Platform/OS you are using along with how you installed ifm3dpy and if possible Please share the sample code That's producing the error.

nikhilnagaraj, 2022-08-01:
@in

> @nikhilnagaraj Can you please specify the Platform/OS you are using along with how you installed ifm3dpy and if possible please share the sample code that's producing the error.

I am running it on WSL (Ubuntu). The sample code is 

```
from ifm3dpy import O3RCamera, FrameGrabber, ImageBuffer
import matplotlib.pyplot as plt

o3r = O3RCamera()
config = o3r.get()
config["ports"]["port0"]["state"] = "RUN"
config["ports"]["port2"]["state"] = "RUN"
o3r.set(config)
fg = FrameGrabber(o3r, pcic_port=50010) #Expecting a head on Port 0 (Port 0 == 50010)
im = ImageBuffer()

if fg.wait_for_frame(im, 1000):
    plt.imsave("saved_image.png", im.distance_image())
```

Installed the python library with `pip install ifm3dpy`.

inbangsa, 2022-08-01:
@nikhilnagaraj

 ```fg = FrameGrabber(o3r, pcic_port=50010)``` can you please change the port to 50012. 

port0->50010-> 2D image
Port2 -> 50012 -> 3D image ( Distance, Amplitude, xyz)

nikhilnagaraj, 2022-08-01:
@inbangsa 

Perfect! That works :) Should have spotted this myself.

Would recommend changing it in the docs and if possible making the error message more descriptive. If you can point me to the appropriate location, I can create a PR for the same.

lola-masson, 2022-08-01:
Hi @nikhilnagaraj,
Happy to accept PRs, thanks for offering. The examples are implemented [here](https://github.com/ifm/ifm3d/tree/v0.93.0/examples/o3r) along with md doc. 
Please keep in mind that the `o3r/main-next` branch is a work in progress and currently contains breaking changes to the API. I recommend pointing to the `v0.93.0` branch in your PR and we will integrate your changes to the next API release.

closed 2022-10-22 by github-actions[bot]