ifm/ifm3d #362: ifm3d::Image to cv::Mat conversion

j-friedl, 2022-08-01:
Hi!
I'm wondering if there's an example on how to convert the `ifm3d::Image` to an `cv::Mat`?
I want to be able to do something like this:

`cv::Mat m = im->DistanceImage();`

It would be really helpful if such conversions make it into the docs :) (also usage with PCL etc.)

inbangsa, 2022-08-01:
@j-friedl

 This  work in progress and should be part of the future release. 

following code will be a good start point : 

```c++
// -*- c++ -*-
/*
 * Copyright 2021-present ifm electronic, gmbh
 * SPDX-License-Identifier: Apache-2.0
 */

#ifndef IFM3D_IMAGE_TO_CV_MAT_HPP
#define IFM3D_IMAGE_TO_CV_MAT_HPP

#include <ifm3d/stlimage/image.h>
#include <opencv2/opencv.hpp>
#include <opencv2/core/mat.hpp>
#include <cstdint>
#include <memory>
#include <map>
  
std::map<uint32_t, int32_t> ifm3d_type_to_cv_type{
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_8U),  CV_8U},
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_8S),  CV_8S},
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_16U), CV_16U},
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_16S), CV_16S},
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_32S), CV_32S},
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_32F), CV_32F},
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_64F), CV_64F},
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_16U2),CV_16U},
  {static_cast<std::uint32_t>(ifm3d::pixel_format::FORMAT_32F3),CV_32F}};

namespace ifm3d
{

  cv::Mat
  convert_to_mat(ifm3d::Image& image)
  {
    int cv_type = CV_MAKETYPE(
      ifm3d_type_to_cv_type[static_cast<std::uint32_t>(image.dataFormat())],
      image.nchannels());

    return cv::Mat(image.height(), image.width(), cv_type, (void*)image.ptr(0));
  }

}// end: namespace ifm3d

#endif // IFM3D_IMAGE_TO_CV_MAT_HPP

```

j-friedl, 2022-08-01:
Great it works thank you!

j-friedl, 2022-08-01:
Yes, thanks!

closed 2022-08-01 by inbangsa