ifm/ifm3d #364: Build the ifm3d from source

piranda85k, 2022-08-02:
Hello, 
I'm struggling to install the ROS 2 driver package, especially with the set-up of ifm3d. 
I've got the same issues as https://github.com/ifm/ifm3d/issues/357#issue-1315071373. 
But even when I download manually the zip code of the project, the errors appear. 
I am working on Ubuntu 20.04.
Does anyone have a solution ?

Thank you for your support.

inbangsa, 2022-08-02:
@piranda85k 

for working with  ifm3d-ros package you need to build the tagged version (0.93.0) of ifm3d.  current o3r/main-next branch is work in progress for  next major release of ifm3d. 

Please use following instruction to build the ifm3d for Ros 

```
 $git clone -b v0.93.0 https://github.com/ifm/ifm3d.git
$ cd ifm3d 
$ mkdir build
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_MODULE_OPENCV=ON -DBUILD_MODULE_PCICCLIENT=ON -DBUILD_MODULE_IMAGE=ON  ..
$ cmake --build .
$ sudo cmake --build . --target install
``

piranda85k, 2022-08-02:
I've tried this several times with the stable compatible versions (0.93.0, 0.92.0, 0.91.0).
And I've got always the same issue : 
[image]
Cannot generate a safe runtime search path for target ifm3d-image-tests
Cannot generate a safe runtime search path for target ifm3d-opencv-tests
Cannot generate a safe runtime search path for target  ifm3d-pcicclient-tests

lola-masson, 2022-08-02:
Hi @piranda85k,
Note that in the latest updates of ifm3d-ros2, we actually removed the dependency to openCV and PCL. So you should be able to build ifm3d from the tagged ifm3d release with the usual cmake instructions:
```
$git clone -b v0.93.0 https://github.com/ifm/ifm3d.git
$ cd ifm3d 
$ mkdir build
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX=/usr  ..
$ cmake --build .
$ sudo cmake --build . --target install
```
Since you had issues with previous installations, I would also double check that there is no version conflicts. Check in /usr/lib/cmake: is there leftover files there from other versions? Remove them. Same in /usr/lib, you can cleanup all the libifm3d*.so files. Also remember to remove the CMakeCache.txt from the build folder or build from an empty build folder. After that try the build again.
You also can check `which ifm3d` and make sure it uses the library in /usr/bin.

piranda85k, 2022-08-03:
I've cleaned all my old files using the command below to locate them : 
``` sudo find / | grep ifm3d\n  ```
And it worked. 
Thanks for your help.

closed 2022-08-03 by piranda85k