ifm/ifm3d #366: Trying to load docker container on O3R VPU and I get "no space left on device"

niallomahony93, 2022-08-20:
Hi ,  I am trying to deploy custom code to the O3R VPU (starting off with just the basics for now but the overall goal is to deploy some lightweight pytorch models for inference on the camera streams)

I followed the steps of this [tutorial] (https://ifm3d.com/sphinx-doc/build/html/documentation/O3R/Docker/deployVPU.html?highlight=ssh) 

I downloaded the docker image provided by ifm with `docker` pull ghcr.io/ifm/ifm3d:latest-l4t@sha256:71a34e14f279a76e25b840b6eb0297a393b09f3ad311b0f51667ef00560bd928`

I used docker save to make it a tar file and transferred it to the vpu via scp

When i run `docker load ` it proceeds loading layers so far until I get

> Error processing tar file(exit status 1): write /home/ifm/venv/lib/python3.9/site-packages/numpy/polynomial/tests/test_printing.py: no space left on device

I inspected the docker image and its size is 1.02GB

df - h returns the following when run of the vpu via ssh:

> o3r-vpu-c0:~$ df -h
Filesystem       Size  Used Avail Use% Mounted on
none             1.6G     0  1.6G   0% /dev
/dev/mmcblk0p1   6.1G  4.2G  1.6G  74% /
tmpfs            1.9G  4.0K  1.9G   1% /dev/shm
tmpfs            1.9G  9.8M  1.9G   1% /run
tmpfs            1.9G     0  1.9G   0% /sys/fs/cgroup
tmpfs            1.9G  4.0K  1.9G   1% /tmp
tmpfs            1.9G   40K  1.9G   1% /var/volatile
/dev/mmcblk0p33  1.2G  9.0M  1.1G   1% /opt/ifm/data
overlay          1.2G  9.0M  1.1G   1% /etc/systemd/network
tmpfs            383M     0  383M   0% /run/user/989

Is there any step I am missing to get a container with ifm3d libraries running on the vpu?

desengph, 2022-08-22:
Hi @niallomahony93 

There seem to be two problems here:
1. The Docker save to tar and scp copy to the VPU and an additional docker load on the VPU takes "double" the space of the container to unpack it. This is a known bug. [Please have a look at this description on how to avoid file size limitations on the VPU](http://ifm3d.com/sphinx-doc/build/html/documentation/O3R/Docker/deployVPU.html?highlight=docker%20load#scp)

```
docker save <image> | ssh -C oem@192.168.0.69 docker load
``` 

2. The VPU space seems to be quite “full” already. If you don't need older Docker containers and Images anymore, please consider a [`docker system prune`](https://docs.docker.com/engine/reference/commandline/system_prune/). This will remove dangling images, not used containers, etc.
If the space is not release directly (after a docker system / container prune), some min or a reboot is sometimes required.

If you want to deploy code using CUDA a specialized firmware is required, which comes with the CUDA libs and dependencies. 
Please get in touch via email at support.robotics@ifm.com for a download link.

niallomahony93, 2022-08-22:
Thank you, I was able to deploy the container with the instructions provided and will be in contact re: firmware for CUDA deployment

closed 2022-08-22 by niallomahony93