ifm/ifm3d #368: Intermittent errors while using `wait_for_frame()`

nikhilnagaraj, 2022-09-28:
**Describe the bug**
I am using the O3R225 - OVP800 combination to capture RGB and Pointcloud data. Ideally, I would like to capture images at random points in time (based on external inputs). However,  the `wait_for_frame` function seems to be highly unreliable. Some of the errors that I have seen while invoking this function are:

* `importlib._bootstrap.Error: No such file or directory`
* `importlib._bootstrap.Error: Invalid argument`
* `importlib._bootstrap.Error: Connection reset by peer`

These errors occur more often while I try to capture data from the 3D port (they occur less frequently on the 2D port).

> The VPU is not directly connected to the system where the scripts are executed but they are on the same network and I am consistently able to ping the device (thus likely eliminating network issues?).

> This problem seems to vanish when I stream frames from the device using the `on_next_frame(...)` function.

**To Reproduce**
A minimal script to reproduce the error is attached below.

**Expected behavior**
`wait_for_frame()` should reliably return a frame.

**Output/Screenshots**
If applicable, add console output and/or screenshots to help explain your problem.

**Minimal Working Example**
```python
import asyncio
from random import random
import time
from ifm3dpy import O3R, FrameGrabber, buffer_id

o3r = O3R()
rgb_fg = FrameGrabber(o3r, 50011)
dist_fg = FrameGrabber(o3r, 50012)
rgb_fg.start([buffer_id.JPEG_IMAGE])
dist_fg.start([buffer_id.NORM_AMPLITUDE_IMAGE, buffer_id.RADIAL_DISTANCE_IMAGE, buffer_id.XYZ])

def reset_ports_config():
    config = o3r.get()
    config["ports"]["port1"]["state"] = "CONF"
    config["ports"]["port2"]["state"] = "CONF"
    o3r.set(config)
    config["ports"]["port1"]["state"] = "RUN"
    config["ports"]["port2"]["state"] = "RUN"
    o3r.set(config)

async def main():
    """Capture data from the sensor
    """
    reset_ports_config()
    while True:
        _ = await rgb_fg.wait_for_frame()
        _ = await dist_fg.wait_for_frame()
        print("Received frame...")
        # Sleep for a random interval from 0 - 5 seconds
        time.sleep(random() * 5)

if __name__ == "__main__":
    asyncio.run(main())

```

**Configuration/Environment (please complete the following information):**
 - OS: `WSL 2, Ubuntu 20.04.4 LTS`
 - Python Version: `3.10.4`
 - IFM3DPY Package Version: `1.0.0`

**Camera Configuration**

```json

{
[... 377 lines cut ...]
          "mixedPixelFilterMode": 1,
          "mixedPixelThresholdRad": 0.05
        },
        "extrinsicHeadToUser": {
          "rotX": 0.0,
          "rotY": 0.0,
          "rotZ": 0.0,
          "transX": 0.0,
          "transY": 0.0,
          "transZ": 0.0
        },
        "version": {
          "major": 0,
          "minor": 11,
          "patch": 12
        }
      },
      "state": "RUN"
    }
  }
}
```

**Additional context**
Add any other context about the problem here.

lola-masson, 2022-09-29:
Hi @nikhilnagaraj,
I just tried your script and have not yet seen any of these errors you are listing. Do you have an idea of how long the script has to run before you see the issues?

The `wait_for_frame()` is not going to perform well in a `while True` loop but I would expect that with the `sleep()` you put in there it should be sufficient to not run into any delay.

Are you able to give us more details on your network connection? Which ethernet port are you using (it looks like eth0 from the json config but just want to confirm)? Are you connected through a router, wifi, etc?

nikhilnagaraj, 2022-09-30:
Hi @lola-masson 
It usually occurs within the first 20-25 calls. I suspected that it was a network issue, but the thrown errors made me think otherwise. 

Regarding the network connection, I am using `eth0`  and the VPU is on the same network as I am (not connected directly). I am able to ping the VPU and receive frames semi-regularly, so I am unsure if this is an issue with the network. However, in a few days, I will be testing this with a physical cable connecting the VPU to my device (or by pushing a docker image to the VPU) so I should have better feedback. 

Might the issue of having to reset the RGB imagers to CONF and then RUN  as listed [here](https://ifm3d.com/sphinx-doc/build/html/documentation/O3R/Firmware/ReleaseNotes/Firmware_0.14.x_Release_Notes.html#known-issues) be related to this? I have definitely seen that I have to reset the RGB imagers to be able to receive any data if the delay between subsequent frame requests is long enough. 
(Similar issue with the distance imagers, where changing the mode to `standard_range2m` and then back to `standard_range4m` seems to help).

lola-masson, 2022-10-03:
It seems like this might be a bug on our end. The fact that you have to "restart" each imager by switching is CONF->RUN or standard_range2m->standard_range4m is strange, this should not happen and we are investigating. We will let you know as soon as we find the issue and provide a solution.

desengph, 2022-10-13:
@nikhilnagaraj can you please contact us via email at: support.robotics@ifm.com

We tried to reach you via your linked email address on your profile on GitHub, but the emails can't be delivered.

nikhilnagaraj, 2022-10-13:
> @nikhilnagaraj can you please contact us via email at: [support.robotics@ifm.com](mailto:support.robotics@ifm.com)
> 
> We tried to reach you via your linked email address on your profile on GitHub, but the emails can't be delivered.

Done! :)

closed 2022-11-28 by github-actions[bot]