ifm/ifm3d #369: Build from source, XMLRPC was not found

piranda85k, 2022-10-05:
Hello, 
I'm trying to build from source the ifm3d package (0.20.3) on my Jetson tx2 to be able to receive the data of my O3D303 camera with a python script.
But I'm struggling to compile from source the package. 

I have the following error message :

```
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
XMLRPC was not found.  Make sure the entries XMLRPC_* are set.  (missing:
XMLRPC_C_FOUND XMLRPC_LIBRARIES)
Call Stack (most recent call first):
/usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
/usr/share/cmake-3.10/Modules/FindXMLRPC.cmake:125 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
modules/camera/src/libifm3d_camera/CMakeLists.txt:10 (find_package)

- - Configuring incomplete, errors occurred!
See also "/home/prototo/ifm3d_ws/ifm3d-0.20.3/build/CMakeFiles/CMakeOutput.log".
See also "/home/prototo/ifm3d_ws/ifm3d-0.20.3/build/CMakeFiles/CMakeError.log".
```

I've checked the issue https://github.com/ifm/ifm3d/issues/127#issue-393578395
But the library **libcurl4-gnutls-dev** is well installed on my computer. 

Have you ever encountered this type of problem? Do you have a solution or an idea where it could come from?

graugans, 2022-10-05:
Hej @piranda85k ,

Are you following the instructions here: https://github.com/ifm/ifm3d/blob/v0.20.3/doc/source_build.md ? What version of JetPack are you using?

piranda85k, 2022-10-05:
My bad, I had not seen this instructions

closed 2022-10-05 by piranda85k