ifm/ifm3d #389: Need SDK (Software developer's Kit) for O3X

wwwwqqqqliao, 2023-03-17:
Dear Mr/Ms,

Could you please do me a favor?

If it is possible, could you please share and send me SDK(Software developer's Kit) for O3X?

Please send me email address: wwwwqqqq_liao@126.com or call me +8615889645978.

Your assistance would be very appreciated.

My best regards!
Liao

wwwwqqqqliao, 2023-03-17:
Could anyone could share me C++ and C# Software developer's Kit? Thx alot

lola-masson, 2023-03-17:
Hi @wwwwqqqqliao,

The API to use with the O3X is the ifm3d library. The official release for the O3X is version 0.20.3. You can find it here: https://github.com/ifm/ifm3d/releases/tag/v0.20.3.

wwwwqqqqliao, 2023-03-17:
hi,
 What I need is an SDK, including. H file and.cpp file. Not the released software

wwwwqqqqliao, 2023-03-17:
hi,
 What I need is an SDK, including. H file and.cpp file. Not the released software
such as  "SDK
ifmO3D3xx PDM SDK 1.22.80
.zip(36.8 MB)
SHA-256",

What I need is the SDK corresponding to O3X, a development kit similar to the above file (ifmO3D3xx PDM SDK 1.22.80)

wwwwqqqqliao, 2023-03-17:
hi.

hello:

I need the same SDK development as in the picture above, which corresponds to the O3D model, but I want the O3X model, thank you

wwwwqqqqliao, 2023-03-18:
> The ifmO3D3xx PDM SDK is deprecated, and support for O3D and O3X platforms migrated to the ifm3d library, in the version 0.20.3 I mentioned above. The code is open source and you can get all the .cpp and .h files directly for GitHub: https://github.com/ifm/ifm3d/tree/v0.20.3.

hi.
  Which folder are the header files, source files, and library files? I git down. But I didn't get what I wanted. Above are the files and folders I downloaded. I can't find them. When we develop our own programs, we usually need an SDK from the manufacturer. Just like the O3D product you have online. You're talking all over the place but you haven't solved the SDK package I asked for. Could you send it to me directly as an attachment? thank you

graugans, 2023-03-20:
Hej @wwwwqqqqliao, from your request I guess the operating system you are working with is Windows? Am I right? To build the SDK with the header and source files you have to follow the instructions [here](https://github.com/ifm/ifm3d/blob/v0.20.3/doc/windows.md) this will provide you the things needed. What Visual Studio Version are you using?

closed 2023-05-05 by github-actions[bot]