ifm/ifm3d #398: gpg: keyserver receive failed: Server indicated a failure

Lyk77, 2023-06-19:
root@61f86a7af7e2:/etc# sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 8AB59D3A2BD7B692
Executing: /tmp/apt-key-gpghome.K5OWoHWrnC/gpg.1.sh --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 8AB59D3A2BD7B692
gpg: keyserver receive failed: Server indicated a failure

lola-masson, 2023-06-20:
Hi @Lyk77,
Could you please give us more details about what you are trying to do here? It is unclear to me whether this error message is related to ifm. Thanks!

DevnathNair, 2023-07-11:
@lola-masson I believe it's related to the keyserver not responding. I'm facing the same issue as well.

`Executing: /tmp/apt-key-gpghome.an4Ls3y7Ab/gpg.1.sh --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 8AB59D3A2BD7B692
gpg: failed to receive from keyserver: Server indicated a failure`

Is there an upated key/keyserver?

closed 2023-08-25 by github-actions[bot]