ifm/ifm3d #405: sntp server cannot modify!!?

AIWildUItraman, 2023-07-07:
我想修改设备的授时服务器，因为设备默认的是谷歌授时服务。在国内无法正确授时。
我使用下面命令修改的：
echo {}  | jq  '.device.clock.sntp.systemClockSynchronized=true'  | ifm3dpy config
echo {}  | jq  '.device.clock.sntp.activeServerName="cn.ntp.org.cn"'  | ifm3dpy config
echo {}  | jq  '.device.clock.sntp.activeServerAddress="223.113.120.195"'  | ifm3dpy config
修改完查看配置，发现还是默认的没有变。
我修改授时服务的目的是因为我ssh 进入设备后，然后进入docker  然后使用apt update 时候无法更新，我需要安装一些包，时间不对无法安装。

graugans, 2023-07-07:
English below:

I want to modify the time service server of the device, because the default time service of the device is Google time service. It is impossible to tell the time correctly in the country.
I modified it with the following command:
echo {} | jq '.device.clock.sntp.systemClockSynchronized=true' | ifm3dpy config
echo {} | jq '.device.clock.sntp.activeServerName="cn.ntp.org.cn"' | ifm3dpy config
echo {} | jq '.device.clock.sntp.activeServerAddress="223.113.120.195"' | ifm3dpy config
After modifying the configuration, I found that the default has not changed.
The purpose of my modifying the time service is because I can’t update when I ssh into the device, then enter docker and then use apt update. I need to install some packages, and the time is wrong and cannot be installed.

graugans, 2023-07-07:
@AIWildUItraman please check the documentation here: https://ifm3d.com/documentation/Technology/VPU/sntp.html?highlight=ntp

Please keep in mind for a correct name resolution you also need to set the correct DNS server. So maybe the IP is preferred. Of course the NTP Server needs to in a subnet the VPU can reach. So in case you are using a SNTP server in the internet the VPU needs internet access. So I recommend a local installation. Please check

AIWildUItraman, 2023-07-11:
OK thanks

closed 2023-07-11 by AIWildUItraman