ifm/ifm3d #406: ifm3dpy

cboostkmokhtar, 2023-07-10:
**Cannot import O3D from 'ifm3dpy'**
From the [ifm3d documentation](https://api.ifm3d.com/stable/examples/o3r/getting_data/getting_data.html ), it says I can use the Python API for my O3D camera (03d301). It did not work for me.

**To Reproduce**
pip install ifm3dpy
from ifm3dpy import O3R (no issues with import)
from ifm3dpy import O3D (Issue see below)

**Expected behavior**
No import errors

**Output/Screenshots**
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Cell In[4], line 1
----> 1 from ifm3dpy import O3D

ImportError: cannot import name 'O3D' from 'ifm3dpy' (/home/kamal/.local/lib/python3.10/site-packages/ifm3dpy.cpython-310-x86_64-linux-gnu.so)

cboostkmokhtar, 2023-07-10:
@desengph I have checked the migration document; it is unclear which ifm3dpy version I should use to acquire data from my O3D camera (03d301).

desengph, 2023-07-11:
@cboostkmokhtar Please see the compatibility matrix [here](https://github.com/ifm/ifm3d#current-revision)

For an O3D device please use ifm3d version 0.20.x

cboostkmokhtar, 2023-07-11:
@desengph That is fine. It is working, I have installed the .deb files in the docker container, and I can run the simple cpp example to acquire the depth and amplitude image.
 
What I am looking for initially in this issue is using ifm3dpy and not ifm3d. So that I can use the camera directly with my Python script.

lola-masson, 2023-07-18:
For ifm3d 0.20.x, you need to build the python bindings from source. Follow the build instructions here: https://github.com/ifm/ifm3d/blob/v0.20.3/doc/python.md.

closed 2023-09-01 by github-actions[bot]