ifm/ifm3d #409: set: RPC response indicates failure.  Invalid device configuration

AIWildUItraman, 2023-07-12:
code :
from ifm3dpy.device import O3R
o3r = O3R(ip='192.168.3.69')
config=o3r.get()
port = 'port0'
config["ports"][port]["state"]="RUN"
print(config["ports"][port]["state"])
o3r.set(config)

Error:
(venv) root@e5617229cd63:/date# /home/ifm/venv/bin/python /date/src/1.py
RUN
2023-07-12 08:49:54 ERROR [/builds/syntron/support/csr/ifm3d/ifm3d/modules/device/src/libifm3d_device/xmlrpc_wrapper.hpp:98] http://192.168.3.69:80/api/rpc/v1/com.ifm.efector/ -> set: RPC response indicates failure.  Invalid device configuration
Traceback (most recent call last):
  File "/date/src/1.py", line 7, in <module>
    o3r.set(config)
ifm3dpy.device.Error: Unknown error 104014: Invalid device configuration

dekhanra, 2023-07-12:
Hi @AIWildUItraman,

Can you send the json config by executing the following command
```sh
$ifm3dpy dump
```

AIWildUItraman, 2023-07-12:
sure! I can successfully ping the device's IP address and also retrieve the configuration using o3r.get(). However, I'm unable to use o3r.set(config) to update the configuration.
My firmware version is the latest, 1.0.14
it don't happen in 0.16.23

desengph, 2023-07-14:
@AIWildUItraman 

There are a couple of possible cases where you can end up in the described error case as above:
1. Setting an invalid JSON
2. Setting a JSON which doesn't match the hardware connectivity
3. A persistent JSON saved on the device from previous operation which doesn't match the current hardware connectivity

## 1. invalid JSON

My best guess is that you're trying to set parameters which might have changes between versions:
For verification purposes, we provide a [JSON schema](https://api.ifm3d.com/stable/_autosummary/ifm3dpy.device.O3R.html#ifm3dpy.device.O3R.get_schema) in addition to the regular configuration JSON.
The verification can be implemented similar to this [example](https://github.com/ifm/documentation/blob/main/ODS/Python/examples/ods_config.py#L30)

## 2.  Setting a JSON which doesn't match the hardware connectivity
If the JSON configuration doesn't match the current hardware connectivity, i.e. you try to set parameters of a RGB stream to a 3D TOF stream, the JSON might still appear as a valid config according to the schema.
But the parameters can not be applied to this port. 

## 3.  A persistent JSON saved on the device from previous operation which doesn't match the current hardware connectivity
If a save_init has been called in the past, the hardware connection is frozen in regard to the connectivity at the time of the method call. If the hardware connections are changed in the meanwhile, i.e. different heads are connected to the same port, the device detects this and shows errors & diagnostic entries.
If a persistent JSON is saved on the device, please remove it via a [factory reset](https://api.ifm3d.com/stable/_autosummary/ifm3dpy.device.O3R.html#ifm3dpy.device.O3R.factory_reset).

For further information, please provide the diagnostics messages during the time of the error call.
Documentation for how to receive them can be found [here](https://ifm3d.com/documentation/SoftwareInterfaces/ifmDiagnostic/index_diagnostic.html). 

Please also include the JSON configuration (retrieved via a get / dump) and also the requested JSON config (the one that fails) as stated before by @dekhanra 

Thanks!

AIWildUItraman, 2023-07-18:
Thank you for your guidance. This issue disappeared after I flashed a new firmware.

closed 2023-07-18 by AIWildUItraman