ifm/ifm3d #413: O3D303 measure dimensions on Ubuntu 22

SystemDiagnosticss, 2023-08-02:
Hi. I have IFM O3D303 camera. I need to measure dimensions of different boxes on a conveyor.  In my case, I want to connect it to Raspberry Pi 4 under Ubuntu 22. I would be satisfied with a ready-to-use solution from Wizard "Dimensioning of rectangular objects" but O3D303 does not support it (it works only on O3D300, O3D302).  And I no have idea how I can run it under Ubuntu. 

So my question is - How I can measure dimensions of boxes using IFM O3D303 under Ubuntu?

lola-masson, 2023-08-02:
Hi @SystemDiagnosticss,

This feature is only available for the O3D302 (the sensor, not the camera). And the algorithms on board the sensor are only available through the Vision Assistant. For Ubuntu and the O3R303, you would need to implement the algorithm yourself.

SystemDiagnosticss, 2023-08-02:
Thanks for your answer @lola-masson, maybe you know about some source code examples for it on Python or C++ ? Maybe someone has already done this?

lola-masson, 2023-08-02:
We don't have any examples ready for object dimensioning. Maybe have a look at [Open3D](http://www.open3d.org/docs/release/index.html) has something that you could use.

SystemDiagnosticss, 2023-08-02:
And last questions. Is it possible to run something like visionassistance.dll and run it without x-screen and use maybe some API to send data from o3d302 with pre-installed "Dimensioning of rectangular objects" via Ethernet connection to my app?

graugans, 2023-08-03:
Since the O3D302 is using the same communication mechanism like the O3D303 this shall be possible with the ifm3d. In case you have a configured Dimensioning app it is just a matter of configuring the result data stream and also to interpret that stream. You can check if the [JSON_MODEL](https://github.com/ifm/ifm3d/commit/2e8dff4fd7ef7bb7bbbf372775fbbd7f84599e00) contains the required result data.

SystemDiagnosticss, 2023-08-03:
@graugans Thanks I try

closed 2023-08-03 by SystemDiagnosticss