ifm/ifm3d #419: Framgegrabber - JSON Path not valid: /applications/instances

mxmahmoud, 2023-09-04:
I have pulled the latest repo and tried to execute "ifm3dpy_viewer.py"

commit 81a99cf79cb0a555e65a0c3b3f98dd34a4a199ba (tag: v1.4.1)
Merge: 78a9081 d0aac1f
Author: Bangadkar, Sachin <sachin.bangadkar@ifm.com>
Date:   Thu Aug 31 14:04:45 2023 +0000

    Merge branch 'Release/v1.4.1' into 'main'

i have handed over the following arguments ip ("172.20.129.43"), pcic port (50010), image="xyz" and left everything else as it is.

the error appeared in the call of FrameGrabber as follows
**Error Message**
```
 cd /home/john_doe/projects/INTERNAL/company_v2 ; /usr/bin/env /home/john_doe/miniconda3/envs/company_test/bin/python /home/john_doe/.vscode/extensions/ms-python.python-2023.14.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher 55193 -- /home/john_doe/projects/INTERNAL/company_v2/app/sensors/ifm/ifm3d/examples/o3r/viewer/ifm3dpy_viewer.py 
2023-09-04 16:40:34 ERROR [/home/john_doe/projects/INTERNAL/company_v2/app/sensors/ifm/ifm3d/modules/device/src/libifm3d_device/xmlrpc_wrapper.hpp:98] http://172.20.129.43:80/api/rpc/v1/com.ifm.efector/ -> get: RPC response indicates failure.  JSON Path not valid: /applications/instances
Traceback (most recent call last):
  File "/home/john_doe/miniconda3/envs/company_test/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/john_doe/miniconda3/envs/company_test/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/john_doe/.vscode/extensions/ms-python.python-2023.14.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/__main__.py", line 39, in <module>
    cli.main()
  File "/home/john_doe/.vscode/extensions/ms-python.python-2023.14.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 430, in main
    run()
  File "/home/john_doe/.vscode/extensions/ms-python.python-2023.14.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 284, in run_file
    runpy.run_path(target, run_name="__main__")
  File "/home/john_doe/.vscode/extensions/ms-python.python-2023.14.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 321, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "/home/john_doe/.vscode/extensions/ms-python.python-2023.14.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 135, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "/home/john_doe/.vscode/extensions/ms-python.python-2023.14.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 124, in _run_code
    exec(code, run_globals)
  File "/home/john_doe/projects/INTERNAL/company_v2/app/sensors/ifm/ifm3d/examples/o3r/viewer/ifm3dpy_viewer.py", line 115, in <module>
    asyncio.run(main())
  File "/home/john_doe/miniconda3/envs/company_test/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/home/john_doe/miniconda3/envs/company_test/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/john_doe/projects/INTERNAL/company_v2/app/sensors/ifm/ifm3d/examples/o3r/viewer/ifm3dpy_viewer.py", line 104, in main
    fg = FrameGrabber(cam, pcic_port=args.pcic_port)
ifm3dpy.device.Error: Unknown error 104012: JSON Path not valid: /applications/instances
```

**Configuration/Environment (please complete the following information):**
 - OS: [e.g. Ubuntu 22.04]
 -  ifm3d version: 
 commit 81a99cf79cb0a555e65a0c3b3f98dd34a4a199ba (tag: v1.4.1)
Merge: 78a9081 d0aac1f
Author: Bangadkar, Sachin <sachin.bangadkar@ifm.com>
Date:   Thu Aug 31 14:04:45 2023 +0000

    Merge branch 'Release/v1.4.1' into 'main'
 - using python

Stedag, 2023-09-05:
Hi @mxmahmoud 

So that I can replicate the issue, which version of the VPU firmware are you using?

Stedag, 2023-09-05:
You will also want to ensure that a 3d head is attached to port0. tested the same script.

Also, ensure that port1 does not have a 2d head connected to it per https://ifm3d.com/documentation/GettingStarted/Unboxing/hw_unboxing.html?highlight=two%20heads%20same%20type

mxmahmoud, 2023-09-06:
I have port 0, 1 connected to 3D heads and port 2,3 to 2D respectivley [0,2] [1,3]
and I am using OVP800 Firmware 1.0.14-973

I have uninstalled the python package which was installed by using the repo and installed ifm3dpy via the "pip install" command without using the repo and that worked for me. I will wait for a stable release, no hurry in this matter. Thank you!

REB427, 2023-09-11:
I am having the same issue ifm3d 1.4.1 (C++), VPU Frimware 1.0.14:
Unknown error 104012: JSON Path not valid: /applications/instances

in my system the exception happens while initializing the error-framegrabber on PCIC Port 50009
std::make_shared<ifm3d::FrameGrabber>(o3R, 50009);

lola-masson, 2023-09-11:
Thanks for reporting the issue. I created a [known issue](https://github.com/ifm/ifm3d/issues/421) as a way to track this and it will be fixed in the next release.

REB427, 2023-10-11:
Problem seems to be solved with version 1.4.3 and can be closed from my side