ifm/ifm3d #420: Wrong inverse intrinsics from O3D305 with ifm3d 1.4.1, C++

sewrkleis, 2023-09-11:
Hi,
with the following code I get strage inverse intrinsics from O3D305 camera
const auto& buffer = img->GetBuffer(ifm3d::buffer_id::INVERSE_INTRINSIC_CALIBRATION);
int width = buffer.width();
int height = buffer.height();
std::vector<double> intr(width*height/sizeof(double), 0.0);
std::memcpy(intr.data(), buffer.ptr(0), width*height);

result:
[0]: 3.9123517191159456e+17
[1]: 858925417549330.25
[2]: -2.7037003746954724e-06
[3]: -8.7463578209951619e-14
[4]: 0
[5]: 0
[6]: 1.6049665430689486e-314
[7]: -3.058777675590818e-10

did you change something compared to older versions < 1.0 of ifm3d or am I doing something wrong?
Thanks

lola-masson, 2023-09-11:
Hi @REB427,

What firmware version are you using on your O3D305?

BigBoot, 2023-09-12:
Hi @REB427, 

with the new version all intrinsic values are exposed instead of just the translations and rotations. 
You should be able to access the values using the deserialize module like this:

```c++
#include <ifm3d/deserialize/deserialize_o3d_buffers.hpp>

auto inv_intrinsics = ifm3d::O3DInstrinsicCalibration::Deserialize(buffer);

float trans_x = inv_intrinsics[ifm3d::intrinsic_param::TRANS_X];
float trans_y = inv_intrinsics[ifm3d::intrinsic_param::TRANS_Y];
float trans_z = inv_intrinsics[ifm3d::intrinsic_param::TRANS_Z];
float rot_x = inv_intrinsics[ifm3d::intrinsic_param::ROT_X];
float rot_y = inv_intrinsics[ifm3d::intrinsic_param::ROT_Y];
float rot_z = inv_intrinsics[ifm3d::intrinsic_param::ROT_Z];
```

REB427, 2023-09-12:
That solved the problem, thanks.
Also figured out that my code from above is working when changing double to float.
In older ifm3d versions the intrinsics were double

closed 2023-09-14 by REB427