ifm/ifm3d #425: O3D303 rotation matrix

yasin333, 2023-10-09:
I use ifm O3D303 tof camera.And i rotate camera around it's x  axis.How can i find real z(depth) value?Should i use rotation matrix formula or intrinsic calibration code?

dekhanra, 2023-10-10:
Hi   @yasin333 , which ifm3d version you are using ?

yasin333, 2023-10-10:
> Hi @yasin333 , which ifm3d version you are using ?

I use ifm3dpy python library and it’s pip version 0.93.0. I dont’t use ifm vision studio.

closed 2023-11-24 by github-actions[bot]

dekhanra, 2023-11-24:
hI @yasin333, The latest ifm3d version has a XYZ image availability and you can use this buffer to get the depth value. Regarding the v0.93.0 please look at this changelog https://github.com/ifm/ifm3d/blob/main/ChangeLog.md#changed-4 to keep the old behaviour of Camera.