ifm/ifm3d #426: missing poky-glibc-i686-vantage-image-armv7ahf-vfp-neon-toolchain-qte-1.8.1.sh file for cross_compiling

ccqedq, 2023-10-11:
can't find poky-glibc-i686-vantage-image-armv7ahf-vfp-neon-toolchain-qte-1.8.1.sh for cross_compiling, how can I get it?

lola-masson, 2023-10-11:
Hi @ccqedq, could you give us more details about what you are trying to do? Which version of ifm3d, are you creating a docker container, what is the destination architecture + OS you are compiling for, what is the error message you are seeing, etc. Thanks.

ccqedq, 2023-10-11:
I want to compile code for arm architecture using x86 LINUX system, and found some tutorial in ifm3d/doc/cross_compiling.md，which using poky-glibc-i686-vantage-image-armv7ahf-vfp-neon-toolchain-qte-1.8.1.sh to perform some operation. but I can't find it in the project.
[image]

ccqedq, 2023-10-11:
I use ifm3d 1.4.3 and compile the code by source instead of docker

graugans, 2023-10-11:
What platform you want to run ifm3d on? Maybe that specific Toolchain is not the one you are looking for?

closed 2023-10-19 by ccqedq