ifm/ifm3d #427: glog issue

ccqedq, 2023-10-19:
I compile the  v0.18.0 code to dynamic library, when I used it in my code, it can't work. finally I found out the issue caused by glog, ifm3d use the glog, and in my code I also use the glog,how can I avoid this issue?

graugans, 2023-10-19:
Any specific reason why you are using v0.18.0?

ccqedq, 2023-10-19:
I am using this version to adapt to the previous code. I see that you have implemented your own log method in the latest version of v1.4.3. If this issue is not present in the latest version, we can update the camera driver code to the latest version. Could you please confirm that?

lola-masson, 2023-10-19:
Indeed, the dependency to glog was removed in ifm3d 1.3.2 (and above), so you should not have the same compatibility issue.

inbangsa, 2023-10-20:
@matthiasbeyer what version of Glog you are using can you use the same version (v0.3.5) as that of ifm3d in your application. 

OR 

You can try to build ifm3d with your Glog version. Provide the installed path of your glog lib to ```-DCMAKE_PREFIX_PATH``` variable while configuring the ifm3d make build. Use ";" as separator to add multiple paths to ```-DCMAKE_PREFIX_PATH```. 

You can try, as @lola-masson suggested, the port application to use the latest version of ifm3d( v1.4.3 which supports legacy devices O3D and O3X, but this release is under experimental release). In the latest release we updated the ifm3d API for grabbing data using ```FrameGrabber``` module and deprecated Image and opencv module and added C++ buffer which can be used with OpenCV and other libraries. Please follow the [migration guide](https://api.ifm3d.com/stable/content/migration_guide/index.html) to update your application.