ifm/ifm3d #447: Framegrabber timeout error with O3R222

robinbansal08, 2024-03-05:
I have VPU (with IFM_Software: 0.16.23-823), connected with 2 O3R222 and 1 VGA resolution O3R camera. As I am testing VGA O3R camera, we are using old firmware for VPU as new firmware does not support VGA O3R camera. All 3 camera are working fine when checked with ifm vision assistant software. But when I try to grab frames using ifm 3d library (v1.5.2), I am getting framegrabber timeout error with O3R222 but not with VGA resolution O3R camera. I am getting warnings(type must be number, but is null) and error (JSON Path not valid: /applications/instances) for both cameras but in case of VGA resolution O3R camera it is able to capture frame but not in case of O3R222 camera. 

```
2024-03-05 05:36:56 ERROR [xmlrpc_wrapper.hpp:100] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/ -> get: RPC response indicates failure.  JSON Path not valid: /applications/instances
2024-03-05 05:36:56 WARN [o3r_impl.hpp:221] XMLRPC ERRO : Unknown error 104012: JSON Path not valid: /applications/instances
2024-03-05 05:36:56 WARN [o3r_impl.hpp:281] JSON: [json.exception.type_error.302] type must be number, but is null
Timeout waiting for camera!
```

Here is the code:

```
void get_cloud(const std::string& o3r_ip, const std::string& port, pcl::PointCloud<pcl::PointXYZ>::Ptr cloud_xyz){
    auto t1 = high_resolution_clock::now();
    auto cam = std::make_shared<ifm3d::O3R>(o3r_ip);
    const auto PCIC_PORT = cam->Port(port).pcic_port;
    auto fg = std::make_shared<ifm3d::FrameGrabber>(cam, PCIC_PORT);

    fg->Start({ifm3d::buffer_id::XYZ});
    fg->SWTrigger();
    auto future = fg->WaitForFrame();
    if (future.wait_for(3s) != std::future_status::ready){
        std::cerr << "Timeout waiting for camera!" << std::endl;
        abort();
    }
    auto frame = future.get();
    auto xyz = frame->GetBuffer(ifm3d::buffer_id::XYZ);
    convert_to_point_cloud(xyz, cloud_xyz);
    fg->Stop();
    auto t2 = high_resolution_clock::now();
    std::cout << "Time to get point cloud: " << duration_cast<milliseconds>(t2 - t1).count() << " ms" << std::endl;
}
```

desengph, 2024-03-05:
@robinbansal08 can you please let us know what hardware connectivity you are running?
Meaning which head is connected to which physical port on the VPU.

**Please include a complete `ifm3d CLI dump` or [`ifm3d API get`](https://api.ifm3d.com/stable/cpp_api/classifm3d_1_1O3R.html#a4aac4d3ceda92e3143ab4626d843189b) output in this issue.**

There are known limitations where 3D VGA heads can not be connected to the same paired deserializer as a O3R222 or O3R225 on the VPU - similar to the 3D TOF and 2D RGB limitations.

robinbansal08, 2024-03-07:
Hardware connection: 

port 0: RGB O3R222 (a)
port 1: RGB  VGA 
port 2: 3D   O3R222 (a)
port 3: 3D   O3R222 (b)
port 4: 3D   VGA

Here, is the ifm3d CLI dump: [o3rdump.json](https://github.com/ifm/ifm3d/files/14522578/o3rdump.json)

All ports are accessible via ifm vision assistant.

desengph, 2024-03-07:
The VGA and regular O3R222 or O3R225 heads can not be connected to the same paired deserializer:
Meaning VGA TOF, VGA TOF on Port [0,1] [2,3] [4,5] would work.

Can you please reconnect your heads using the following connectivity:
port 0: 3D VGA
port 1: no connection
port 2: 3D O3R222 (a)
port 3: 3D O3R222 (b)
Please disconnect the VGA RGB imagers for test purposes. 

Please let us know if this changes the connectivity.

If errors still persist, can you please send the complete systemD journalctl output (can be accessed on the VPU embedded device as oem user) via email at support.robotics@ifm.com.

Thanks.

robinbansal08, 2024-03-07:
As per that, current connection should work as nothing is connected in pair with 3D-VGA (port-5 is empty).
[0 1] 2D, 2D
[2 3] 3D, 3D (both O3R222)
[4 5] 3D-VGA, no connection

desengph, 2024-03-07:
There are two hypotheses to be double-checked:
- [x] No VGA is connected to the same deserializer as a 2D RGB stream and 38k TOF streams
- [ ] Disconnect the VGA RGB stream completely - since it may be missing the required calibration on the head.

As mentioned before, can you send a systemD journal output to us via email?

robinbansal08, 2024-03-18:
Testing further I found that this issue occurs when camera is set to "IDLE" mode and if we send software trigger to capture frames, it fails with timeout error. But when it is set to "RUN" mode, and we do not send any software trigger, then we are able to get frames. This behavior is experienced with all ports (2D ports and 3D ports of both type of cameras).

ifm-csr, 2024-03-18:
This is probably due to a known instability for the software trigger. Consider the following:
- No trigger is available for the RGB cameras. They will always be "free-running". 
- For the 3D cameras, you might need to send a trigger signal a couple times to be able to get a frame. Could you try this and confirm whether or not you are able to get a frame when in "IDLE" mode using the trigger multiple times?
- Also note that software trigger support was added in FW 0.16.23, but with known limitations (see https://ifm3d.com/latest/Firmware/ReleaseNotes/FW_0.16.x/FW_0.16.x_Release_Notes.html#known-issues). Also due to the prototype nature of the VGA, the trigger for these types of heads is expected to be unstable.

robinbansal08, 2024-03-18:
Sending multiple trigger signals solves the issue for O3R222 but it not reliable in case of VGA cameras and sometimes it fails with timeout error. As a workaround we can set the camera to "RUN" mode which is sufficient for initial testing phase for us. Thanks for the solution.

I have a question regarding temporal filter: does that works also in "IDLE" mode, when we try to capture a frame by sending a software trigger ? If yes, which frames would be used for temporal filter, will it capture multiple new frames first for filtering.

lola-masson, 2024-03-19:
So, we recommend disabling the temporal when using triggering. The filter will keep doing its filtering job on the frames that are collected, but it does not make sense if the scene changes between triggers, because the filter would use completely unrelated frames.

robinbansal08, 2024-03-21:
Thank you for your support. Closing this issue.

closed 2024-03-21 by robinbansal08