ifm/ifm3d #449: Add shlibs files to debian packages

mgrrx, 2024-05-24:
When building the ifm3d ROS package using bloom and sbuild the build fails when runing dpkg-shlibdeps:
```
dpkg-shlibdeps: error: no dependency information found for /usr/lib/libifm3d_framegrabber.so.1.4.3 (used by debian/ros-noetic-ifm3d-ros-driver/opt/ros/noetic/lib/libifm3d_ros.so)
Hint: check if the library actually comes from a package.
dh_shlibdeps: error: dpkg-shlibdeps -Tdebian/ros-noetic-ifm3d-ros-driver.substvars -l/<<PKGBUILDDIR>>/debian/ros-noetic-ifm3d-ros-driver//opt/ros/noetic/lib/ debian/ros-noetic-ifm3d-ros-driver/opt/ros/noetic/lib/libregistration.so debian/ros-noetic-ifm3d-ros-driver/opt/ros/noetic/lib/libifm3d_ros.so returned exit code 2
```
The reason for this is how the ifm3d debian packages were built in first place: They come without a shlibs file in the debian folder.
I tried adding https://cmake.org/cmake/help/v3.6/module/CPackDeb.html#variable:CPACK_DEBIAN_PACKAGE_SHLIBDEPS to the CMakeLists.txt and tried building the package but unfortunately without success:

```
CMake Error at /usr/share/cmake-3.20/Modules/Internal/CPack/CPackDeb.cmake:340 (message):
  CPackDeb: dpkg-shlibdeps: 'dpkg-shlibdeps: error: cannot find library
  libifm3d_common.so.1.5.3 needed by ./usr/lib/libifm3d_device.so.1.5.3 (ELF
  format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
 
  dpkg-shlibdeps: error: cannot continue due to the error above
 
  Note: libraries are not searched in other binary packages that do not have
  any shlibs or symbols file.
 
  To help dpkg-shlibdeps find private libraries, you might need to use -l.
 
  ';
 
  executed command: '/usr/bin/dpkg-shlibdeps --ignore-missing-info -O
  ./usr/lib/libifm3d_device.so.1.5.3';
 
  found files: './usr/lib/libifm3d_device.so.1.5.3: ELF 64-bit LSB shared
  object, x86-64, version 1 (GNU/Linux), dynamically linked,
  BuildID[sha1]=49337f69e0c84353f0838dbb2afc70fd5d61ac96, not stripped
 
  ';
```

I would recommend to change the way how debian packages are built: Create a debian folder with a proper control file and a rules making use of debhelpers cmake build system. This should be straight forward and avoids the above mentioned issue. If there is interest in that I can try providing a PR.

desengph, 2024-05-27:
@mgrrx thank you very much for your proposal.
We will discuss this and let you know if this will be implemented into the debian package creation.

closed 2024-07-11 by github-actions[bot]

desengph, 2024-07-15:
Good morning @mgrrx, we are happy to receive and review the mentioned PR.