ifm/ifm3d #455: libpthread.so.0 DSO missing from command line

robinbansal08, 2024-07-18:
I am trying to build from source ifm3d library version 1.5.3, with options BUILD_SHARED_LIBS=OFF, BUILD_IN_DEPS=OFF.  The build process is failing.
OS: Ubuntu 20.04 

```
$ cmake -DBUILD_SHARED_LIBS=OFF -DBUILD_IN_DEPS=OFF ..
-- Setting build type to 'RelWithDebInfo' as none was specified.
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Module support is disabled.
-- Version: 8.0.1
-- Build type: RelWithDebInfo
-- CXX_STANDARD: 14
-- Performing Test has_std_14_flag
-- Performing Test has_std_14_flag - Success
-- Performing Test has_std_1y_flag
-- Performing Test has_std_1y_flag - Success
-- Performing Test SUPPORTS_USER_DEFINED_LITERALS
-- Performing Test SUPPORTS_USER_DEFINED_LITERALS - Success
-- Performing Test FMT_HAS_VARIANT
-- Performing Test FMT_HAS_VARIANT - Success
-- Required features: cxx_variadic_templates
-- Looking for strtod_l
-- Looking for strtod_l - not found
CMake Warning at modules/common/CMakeLists.txt:21 (message):
  Skipping common unit tests!

-- Found XMLRPC: 1  
-- Found CURL: /usr/local/lib/cmake/CURL/CURLConfig.cmake (found version "7.78.0-DEV")  
CMake Warning at modules/device/CMakeLists.txt:21 (message):
  Skipping libifm3d-device unit tests!

CMake Warning at modules/framegrabber/CMakeLists.txt:13 (message):
  Skipping libifm3d-framegrabber unit tests!

CMake Warning at modules/swupdater/CMakeLists.txt:15 (message):
  Skipping libifm3d-swupdater unit tests!

CMake Warning at modules/deserialize/CMakeLists.txt:23 (message):
  Skipping deserialize unit tests!

-- Configuring done
-- Generating done

$ make -j4
[  1%] Building CXX object _deps/fmt-build/CMakeFiles/fmt.dir/src/format.cc.o
[  3%] Building CXX object modules/common/src/libifm3d_common/CMakeFiles/ifm3d_common.dir/logging/logger.cpp.o
[  5%] Building CXX object _deps/fmt-build/CMakeFiles/fmt.dir/src/os.cc.o
[  7%] Linking CXX static library libfmt.a
[  7%] Built target fmt
[  8%] Linking CXX static library libifm3d_common.a
[  8%] Built target ifm3d_common
[ 10%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/device.cpp.o
[ 12%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/err.cpp.o
[ 14%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/legacy_device.cpp.o
[ 16%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/o3d.cpp.o
[ 17%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/o3r.cpp.o
[ 19%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/o3x.cpp.o
[ 21%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/semver.cpp.o
[ 23%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/util.cpp.o
[ 25%] Building CXX object modules/device/src/libifm3d_device/CMakeFiles/ifm3d_device.dir/version.cpp.o
[ 26%] Linking CXX static library libifm3d_device.a
[ 26%] Built target ifm3d_device
[ 28%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/buffer.cpp.o
[ 30%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/distance_image_info.cpp.o
[ 32%] Building CXX object modules/swupdater/src/libifm3d_swupdater/CMakeFiles/ifm3d_swupdater.dir/swupdater.cpp.o
[ 33%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/frame.cpp.o
[ 35%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/frame_grabber.cpp.o
[ 37%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/o3d_organizer.cpp.o
[ 39%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/o3r_organizer.cpp.o
[ 41%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/o3r_organizer3D.cpp.o
[ 42%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/o3x_organizer.cpp.o
[ 44%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/organizer_utils.cpp.o
[ 46%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/schema.cpp.o
[ 48%] Building CXX object modules/framegrabber/src/libifm3d_framegrabber/CMakeFiles/ifm3d_framegrabber.dir/algo/o3r_uncompress_di.cpp.o
[ 50%] Linking CXX static library libifm3d_framegrabber.a
[ 50%] Built target ifm3d_framegrabber
[ 51%] Linking CXX static library libifm3d_swupdater.a
[ 51%] Built target ifm3d_swupdater
[ 53%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/app_types_app.cpp.o
[ 55%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/cmdline_app.cpp.o
[ 57%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/config_app.cpp.o
[ 58%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/cp_app.cpp.o
[ 60%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/diagnostic_app.cpp.o
[ 62%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/discover_app.cpp.o
[ 64%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/dump_app.cpp.o
[ 66%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/export_app.cpp.o
[ 67%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/imager_types_app.cpp.o
[ 69%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/import_app.cpp.o
[ 71%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/jsonschema_app.cpp.o
[ 73%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/ls_app.cpp.o
[ 75%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/make_app.cpp.o
[ 76%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/passwd_app.cpp.o
[ 78%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/reboot_app.cpp.o
[ 80%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/reset_app.cpp.o
[ 82%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/rm_app.cpp.o
[ 83%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/time_app.cpp.o
[ 85%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/trace_app.cpp.o
[ 87%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/fg/fg_app.cpp.o
[ 89%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/fg/hz_app.cpp.o
[ 91%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/fg/jitter_app.cpp.o
[ 92%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/fg/schema_app.cpp.o
[ 94%] Building CXX object modules/tools/src/libifm3d_tools/CMakeFiles/ifm3d_tools_shared.dir/swupdater/swupdate_app.cpp.o
[ 96%] Linking CXX static library libifm3d_tools.a
[ 96%] Built target ifm3d_tools_shared
[ 98%] Building CXX object modules/tools/src/bin/CMakeFiles/ifm3d.dir/ifm3d.cpp.o
[100%] Linking CXX executable ifm3d
/usr/bin/ld: ../../../framegrabber/src/libifm3d_framegrabber/libifm3d_framegrabber.a(frame_grabber.cpp.o): undefined reference to symbol 'pthread_condattr_setclock@@GLIBC_2.3.3'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [modules/tools/src/bin/CMakeFiles/ifm3d.dir/build.make:106: modules/tools/src/bin/ifm3d] Error 1
make[1]: *** [CMakeFiles/Makefile2:632: modules/tools/src/bin/CMakeFiles/ifm3d.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

```

Interestingly it builds successfully for all other options:
- BUILD_SHARED_LIBS=OFF, BUILD_IN_DEPS=ON.
- BUILD_SHARED_LIBS=ON, BUILD_IN_DEPS=OFF.
- BUILD_SHARED_LIBS=ON, BUILD_IN_DEPS=ON.

graugans, 2024-07-18:
Quick question, what do you want to achieve? When I was building ifm3d inside Yocto, this worked for me:

```
EXTRA_OECMAKE:append = " \
                        -DBUILD_IN_DEPS=OFF \
                        -DFETCHCONTENT_FULLY_DISCONNECTED=OFF \
                        -DBUILD_MODULE_TOOLS=OFF \
                        -DBUILD_MODULE_SWUPDATER=OFF \
"

But keep in mind that this required me to provide xml-rpc and curl from the rest of the build system:

```
DEPENDS = " \
           xmlrpc-c \
           curl \
"
```

robinbansal08, 2024-07-18:
We are trying to build our application which will be deployed as docker container to our system. We want to optimize the size of the container to minimum possible and for that purpose we use static link libraries. Before in our use case we were using O3D303 cameras, for which we were using ifm3d library version v0.20.3 but now we are also adding some new O3R cameras also. That is why we have now moved to ifm3d library version 1.5.3. But we have noticed the application binary (using ifm3d version 1.5.3, built with options BUILD_SHARED_LIBS=OFF, BUILD_IN_DEPS=ON)  has the size almost twice as before (application using ifm3d v0.20.3). That is why we wanted to check the application binary size if we use ifm3d library with different option  BUILD_SHARED_LIBS=OFF, BUILD_IN_DEPS=OFF.  

I have already installed libxmlrpc-c++8-dev and curl in the container.

graugans, 2024-07-18:
I guess you are already using multistage builds to reduce the size of the docker container. as mentioned above you can also reduce the amount of modules build. This may depend on the use-case you are aiming for.

robinbansal08, 2024-07-19:
I am already using multistage builds and building selective ifm3d modules which are required. 

I want to raise one more issue with ifm3d v1.5.3. I am able to build ifm3d library successfully with options  **BUILD_SHARED_LIBS=OFF, BUILD_IN_DEPS=ON** but when including it in the project, I am facing errors related to xmlrpc and curl on which ifm3d is dependent. I believe with BUILD_IN_DEPS=ON, this would have been already be included with ifm3d build.

```
ENV IFM3D_VERSION v1.5.3
RUN git clone  --branch ${IFM3D_VERSION} https://github.com/ifm/ifm3d.git && \
    cd ifm3d && mkdir build && cd build && \
    cmake -GNinja \
    -DBUILD_MODULE_DESERIALIZE=OFF \
    -DBUILD_MODULE_SWUPDATER=OFF \
    -DBUILD_SDK_PKG=ON \
    -DBUILD_MODULE_TOOLS=OFF \
    -DBUILD_SHARED_LIBS=OFF \
    -DCMAKE_INSTALL_PREFIX=/usr .. && \
    ninja && \
    ninja install && \
    cd ../../ && \
    rm -rf ifm3d
```
The error I am facing during building of the project.
```
153.9 [ 50%] Linking CXX executable mv_app
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_framegrabber.a(frame_grabber.cpp.o): in function `ifm3d::FrameGrabber::Impl::AsyncErrorHandler()':
156.6 /ifm3d/build/../modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp:736: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /ifm3d/build/../modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp:736: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_framegrabber.a(frame_grabber.cpp.o): in function `ifm3d::FrameGrabber::Impl::AsyncNotificationHandler()':
156.6 /ifm3d/build/../modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp:766: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /ifm3d/build/../modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp:766: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_framegrabber.a(frame_grabber.cpp.o): in function `ifm3d::FrameGrabber::Impl::~Impl()':
156.6 /ifm3d/build/../modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp:260: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_framegrabber.a(frame_grabber.cpp.o):/ifm3d/build/../modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp:264: more undefined references to `ifm3d::Logger::Get()' follow
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_device.a(device.cpp.o): in function `isV1SWUpdate(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
156.6 /ifm3d/build/../modules/device/src/libifm3d_device/device.cpp:386: undefined reference to `curl_easy_init'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device.cpp:390: undefined reference to `curl_easy_setopt'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device.cpp:391: undefined reference to `curl_easy_setopt'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device.cpp:393: undefined reference to `curl_easy_setopt'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device.cpp:396: undefined reference to `curl_easy_setopt'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device.cpp:400: undefined reference to `curl_easy_perform'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device.cpp:410: undefined reference to `curl_easy_cleanup'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device.cpp:403: undefined reference to `curl_easy_getinfo'
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_device.a(device.cpp.o): in function `ifm3d::Device::Impl::Impl(std::shared_ptr<ifm3d::XMLRPCWrapper>)':
156.6 /ifm3d/build/../modules/device/src/libifm3d_device/device_impl.hpp:61: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device_impl.hpp:61: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device_impl.hpp:66: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /ifm3d/build/../modules/device/src/libifm3d_device/device_impl.hpp:66: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_device.a(device.cpp.o): in function `ifm3d::Device::Impl::~Impl()':
156.6 /ifm3d/build/../modules/device/src/libifm3d_device/device_impl.hpp:69: undefined reference to `ifm3d::Logger::Get()'
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_device.a(device.cpp.o):/ifm3d/build/../modules/device/src/libifm3d_device/device_impl.hpp:69: more undefined references to `ifm3d::Logger::Get()' follow
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_device.a(device.cpp.o): in function `ifm3d::XMLRPCWrapper::XMLRPCWrapper(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short)':
156.6 /ifm3d/build/../modules/device/src/libifm3d_device/xmlrpc_wrapper.hpp:39: undefined reference to `xmlrpc_c::clientPtr::clientPtr()'
[... cut ...]
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_device.a(device.cpp.o): in function `ifm3d::Device::Impl::DeviceParameter(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
156.6 /ifm3d/build/../modules/device/src/libifm3d_device/device_impl.hpp:111: undefined reference to `xmlrpc_c::value::~value()'
156.6 /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libifm3d_device.a(device.cpp.o): in function `xmlrpc_c::value_string::~value_string()':

```

lola-masson, 2024-07-22:
I can confirm that I can reproduce this error. Let us check what a workaround is and we will get back to you.

closed 2024-09-05 by github-actions[bot]