ifm/ifm3d #456: find_package fails to find ifm3d

robinbansal08, 2024-07-18:
I built the ifm3d library version 1.6.0 from source with the option BUILD_SHARED_LIBS=OFF, and it was built successfully. However, in my C++ project, when I tried to include the ifm3d library, it failed to find it.

OS: Ubuntu 20.04

```find_package(ifm3d 1.6.0 REQUIRED)```
 
```
1.236 CMake Error at CMakeLists.txt:28 (find_package):
1.236   By not providing "Findifm3d.cmake" in CMAKE_MODULE_PATH this project has
1.236   asked CMake to find a package configuration file provided by "ifm3d", but
1.236   CMake did not find one.
1.236 
1.236   Could not find a package configuration file provided by "ifm3d" (requested
1.236   version 1.6.0) with any of the following names:
1.236 
1.236     ifm3dConfig.cmake
1.236     ifm3d-config.cmake
1.236 
1.236   Add the installation prefix of "ifm3d" to CMAKE_PREFIX_PATH or set
1.236   "ifm3d_DIR" to a directory containing one of the above files.  If "ifm3d"
1.236   provides a separate development package or SDK, be sure it has been
1.236   installed.
1.236 
1.236 
1.236 -- Configuring incomplete, errors occurred!

```

This issue does not occur if ifm3d library is built with the option BUILD_SHARED_LIBS=ON.

lola-masson, 2024-07-19:
Hi @robinbansal08, I can confirm that I can reproduce this. 

And actually, when building the library with BUILD_SHARED_LIBS=ON, I have a similar issue, when building [our examples](https://github.com/ifm/ifm3d-examples):
```
0.281 CMake Error at ods/CMakeLists.txt:44 (add_executable):
0.281   Target "ods_get_data" links to target "ifm3d::framegrabber" but the target
0.281   was not found.  Perhaps a find_package() call is missing for an IMPORTED
0.281   target, or an ALIAS target is missing?
```

We will work on fixing this.

In the meantime, please use the official release of ifm3d, the latest right now is 1.5.3. The 1.6.0 is a pre-released and (as you noticed) not fully tested.

robinbansal08, 2024-07-19:
Thank you for checking the issue. I have already tried version 1.5.3 and faced some issues (https://github.com/ifm/ifm3d/issues/455) because of which I tried newer version 1.6.0.

closed 2024-09-02 by github-actions[bot]