ifm/ifm3d #472: Issues with connecting to the XML-RPC protocol

archay0, 2024-12-15:
**Describe the bug**
I am trying to read the 3d image from the sensor to generate a point cloud in my 3d shader environment. I can connect to the VPU and get the config out, set my name on the camera or device, view images in the VisionAssistant, and initialize connection to the VPU or camera through code. The issue comes when I start framegrabbing. To connect I have to set a temporary IP address, even though I allowed direct TCP connections to the IP from my machine, to the specific IP both temporary and initial. Then after connecting using the steps in the documentation, I get the correct pcic port and the correct ip, then start the framegrabber, here is what I get -
```
Device initialized successfully...
Device name: archay's O3R
Device family: device_family.O3R
2024-12-15 13:14:18 ERROR [C:\builds\syntron\support\csr\ifm3d\ifm3d\modules\device\src\libifm3d_device\xmlrpc_wrapper.hpp:100] http://169.254.77.133:80/api/rpc/v1/com.ifm.efector/ -> get: RPC response indicates failure.  JSON Path not valid: /applications/instances
2024-12-15 13:14:18 WARN [C:\builds\syntron\support\csr\ifm3d\ifm3d\modules\device\src\libifm3d_device\o3r_impl.hpp:221] XMLRPC ERRO : Unknown error: JSON Path not valid: /applications/instances    
True
```
**To Reproduce**
```
device = initialize_device(ip='169.254.77.133', xmlrpc_port=80)
fg = FrameGrabber(device, pcic_port=50012)
fg.start()
print(fg.is_running())
```

**Expected behavior**
Frames galore?

**Configuration/Environment (please complete the following information):**
 - OS: Win11
 -  ifm3d version: latest
 - using ROS: nah

archay0, 2024-12-15:
fixed it, would like some explanation on the data type ifm3d_ndarray

closed 2024-12-15 by archay0

graugans, 2024-12-16:
Hej @archay0, from the fact you are using a temporary IP address `169.254.77.133` I do assume you have configured your PC or Laptop to DHCP and perform a 1:1 connection between your PC and the VPU. This is intended for testing purposes only. The recommended way is to use a static IP configuration or to use a DHCP Server to provide an IP Address for both your device and the VPU. The benefit of an static IP is the fact you are under full control and do not have to fiddle around with DHCP lease times. 

The error above is related to the fact that you probably have not created any application. `/applications/instances` which is not required for a camera only use-case.

In regards of the `ifm3d_ndarray` data type it depends on the data you request from the device. What data are you interested in?

archay0, 2024-12-18:
thank you for your reply! @graugans i will look into the DHCP server method. For now a normal 1:1 connection after assigning a temporary ip using the vision assistant, since its the same every time. I am interested in the Radial distance image, i got the ifm3d_ndarray as a result which i am converting to a numpy array to send over my flask server. 172x224 array of depth values, to a 3d processing library to instance points at each point after calculating the distance from the source. I just want to know more about this datatype to optimise my process, if that makes sense. I am still new to networking.